ccrisan/motionpie #137: Downloading zipped images

jan-r, 2015-03-10:
Hello, 

I'm having problems with downloading zipped still images. If I browse the images in motionEye and click on the "zipped" button, I get a "busy" response (turning circle). After some time, the motionEye UI is responsive again, but I didn't receive a zip file or get some further notification what to do now. I also browsed the sdcard with ssh, but didn't find anything.

If I'm just doing something wrong, I'll gladly document this feature in the wiki if someone can tell me how it's supposed to work...  :-)

ccrisan, 2015-03-10:
You probably get a timeout of some sort that is not shown as an error message. You most likely have a big number of pictures and your PI is not fast enough to zip them within 5 minutes. I'll make sure to improve the error messages that pop up when these timeouts occur.

jan-r, 2015-03-10:
Yes, it feels like the poor A+ (not overclocked) is at its limit. Number of images was around 800, size 1280x720. It didn't take 5 minutes though, rather 2-3 minutes, until the UI returned to "normal mode". 

In general, accessing pictures over smb or the motioneye image browser seems to bring the A+ to its knees. Sometimes it really seems to lock up so that I have to reboot after some minutes. Do you think this might be a RAM issue?

ccrisan, 2015-03-10:
It can be either the CPU or the limited memory. Zipping 800 pictures can take a long time.

jan-r, 2015-03-10:
Especially when motion is still active in the background, eating up most of the cpu resources. But I don't know if it would in general be accepted if the motion service was shut down during zipping. 

What about the Raspi 2, do you make use of the extra cores? Having motion/camera access separated from networking and web interface should make the device much more responsive.

ccrisan, 2015-03-10:
It takes for sure advantage of the multicore platform. It's the kernel's scheduler that manages the active processes so we get it "for free".

jan-r, 2015-03-10:
I've just checked my running system. Under normal conditions, free -m reports 33 MB of free RAM. 128 MB are configured to be used by the GPU, 88 MB are used by motionPie. When copying images over smb, free RAM drops dramatically below 20 MB and the file transfer starts to stutter. I guess that's why zipping fails.

I guess I need to get a Pi 2 for my cam...

ccrisan, 2015-03-14:
The current zip mechanism loads the entire contents into memory. Zip files larger than a few tens of megabytes are very likely to produce an OOM. I am planning to rewrite this mechanism to read the content directly from the disk rather than loading it entirely into memory, but it's not going to happen any time soon.

On the other hand I enhanced the UI code so that errors appearing during the zip file generation will show up in the UI.

jan-r, 2015-03-15:
Sounds good. I've upgraded my cam to a Raspberry Pi 2 now, which greatly increases the overall performance. Although the recording capabilities are pretty much the same, interacting with the web UI and other network services is much faster now. The Raspberry Pi A+ worked good as a basis for a pure streaming cam, but motion detection, still image saving and network access was a bit too much.

ccrisan, 2015-04-12:
I'll close this issue with the remark that you need a powerful enough board to handle the relatively large number of zipped pictures (in terms of memory and CPU).

closed 2015-04-12 by ccrisan