ccrisan/motionpie #138: Motion detection on demand

vrolland78, 2015-03-11:
Hi,

First of all thank you for your great work, it's a fantastic way to recycle my "old" raspberry pi model B :-)
Everything works fine for me, but i'd like to ask you a new feature.
I will wish to activate the motion detection on demand via a url, activation by horraire range is too restrictive. But maybe this is already possible ?

regards

Vincent

vrolland78, 2015-03-17:
Maybe that my question is understandable, so I'll be simpler: is there a simple way to remotly start motion ??
regards
Vincent

ccrisan, 2015-03-17:
It's not possible yet, but I'm planning on implementing such a feature as soon as I find some spare time.

umbex, 2015-08-29:
looking for same feature
+1