ccrisan/motionpie #142: Allow recording by GPIO

jujuned, 2015-03-15:
Hello,

Motionpie already have sheudule options, but it may be very helpful if we can activate the system by a RF remote, for example...

Can we configure an input pin in GPIO that turn the system On and Off ?

Maybe for a future release...

ccrisan, 2015-03-15:
This will be possible in a future release.

jujuned, 2015-03-15:
Hello and thank you Ccrisan for your fast replies...

Another suggestion :

Sending an alert by SMS, of course with a USB SIM reader...

I know it is not ready easy, but your program offers so much capabilities...

Julien

Samfox2, 2015-04-10:
Activation via GPIO pin (besides motion) would be very nice so it could also be used as a nice door cam.
Push a button and the camera turns on and (optionally) starts a script, e.g. send a message/jpg to a mobile phone.

neftronics, 2015-09-02:
I have been trying to use a GPIO to perform a controlled shutdown of a RPi model A+ using python and the RPi.GPIO module. This is partially successful. I can toggle I/Os and read them, but cannot attach an interrupt to a GPIO like so: "gpio.add_event_detect(7, gpio.FALLING, callback=shutdown)"
This gives the error "runtimeerror failed to add edge detection". This works fine on Raspbian, is there something I need which is not compiled-in with motionpie?
If this can be made to work, then the original poster (and others in this thread) could use this same approach.

ccrisan, 2015-09-02:
Problems with RPi.GPIO have been reported on motionPie. As far as I know, it doesn't work with PI2 at all. I wasn't aware of this particular problem but it may be caused by the same root problem. Any help or suggestion is appreciated.

neftronics, 2015-09-04:
Thanks for the reply. I have very little experience with Linux programming, I am more of a hardware type, only dabbling with the software, so probably can't help very much
.
I tried running Python in verbose (-vv) mode, but the actual runtime error given was the same - no extra info. It did however show me where it looked for (and found) the modules as I imported them. 
I was wondering if there is a particular kernel module which RPi.GPIO relies upon for interrupts. Perhaps this has not been compiled in, or loaded? Could I simply copy the modules over from Raspbian, which is working? It seems this is the only hurdle in the way of achieving most of the wishes in this thread.
I am very impressed with Motionpie by the way, it works very well with minimal setup. I think it is a great project!

villamarinella, 2015-11-30:
Hi all,
I solve the problem with GPIO or with a radio on 443 MHZ by using a second Raspberry PI.
Motion sends me a message on ftp and read it and switch GPIO or a lamp in my bedroom.
I am 67 and my daughter is always scared about me. So I installed cameras in "every" room and one Raspberry is listening to telegram. With this my daughter can switch on different lights to check Papa:-)
Have a nice day

Klaus Werner
Germany