ccrisan/motionpie #143: Overclocking RPI 2

dandymon, 2015-03-15:
Overclocking causes crashes with RPI 2, 
It only has options for the RPI model B, since the RPI 2 already runs at 900MHz.

Maybe consider giving a warning about overclocking, since 4 cores will generate some super heat!

ccrisan, 2015-03-15:
What only has options for the RPI model B?

dandymon, 2015-03-15:
Hello there

I've got a rpi 2, an under the section where you can overclock, you can go
from none (700Mhz) up to Turbo(1000Mhz).
The rpi2 runs at 900Mhz by standard and doesn't seem to cause any problems
of you leave this setting alone.
Changing this setting to the faster option causes it to crash.
Also, the rpi2 reportedly can be over locked to over 1100Mhz (although heat
sinks are needed)

Thanks for making such awesome software, and for free: you sir will get
into heaven.
I'd consider a donate button on got hub if I were you.
Daniel

[image: image1.PNG]

Sent from Daniel's iPhone

ccrisan, 2015-03-15:
@dandymon thanks for your kind words. Actually the donate button is a good idea :) I'll see how one can get such a button on github.

ccrisan, 2015-03-31:
The overclocking setting is under Expert Settings, which should keep unexperienced users away.

closed 2015-03-31 by ccrisan