ccrisan/motionpie #144: motion-mmal broken in 20150315-release

Lumrytioz, 2015-03-16:
Hi ccrisan and thank you for a wonderful job. (Im willing to donate).
I think something broke the motion-mmal support in the 20150315-release. Im running motionpie-raspberrypi2-20150315.img.gz

motioneye.log

/usr/bin/motion-mmal: can't load library 'libbcm_host.so'
Traceback (most recent call last):
  File "/programs/motioneye/motioneye.py", line 362, in <module>
    _start_motion()
  File "/programs/motioneye/motioneye.py", line 322, in _start_motion
    motionctl.start()
  File "/programs/motioneye/src/motionctl.py", line 125, in start
    raise Exception('motion failed to start')
Exception: motion failed to start

Keep up the good work.
/Carl

ccrisan, 2015-03-16:
I accidentaly left out a library path setting. It'll be fixed soon. Thanks for reporting the problem.

joedupa, 2015-05-17:
this is still broken when building motionPie from the source:

[root@mp-bde41f79 ~]# /usr/bin/motion-mmal 
/usr/bin/motion-mmal: error while loading shared libraries: libbcm_host.so: cannot open shared object file: No such file or directory

ccrisan, 2015-05-18:
What exact commit did you build?

joedupa, 2015-05-18:
the one from yesterday, a couple of hours after you announced the new version.
git clone --depth 1 https://github.com/ccrisan/motionPie.git

ccrisan, 2015-05-18:
Thanks for the info. I'll look into it.

rleyden559, 2015-06-19:
I had exactly the same error on the 6/1 release and the 5/24 release.  I went back to 3/17 release where the issue was explicitly fixed.

ccrisan, 2015-06-21:
@rleyden559 are you trying to build from source or are you talking about the prebuilt image?

rleyden559, 2015-06-22:
Hello Calin

 

I was trying pre-built images.  I tried release 6-21 and 5-24 before going
all the way back to 3-17.  In each case setting the motion executable to
motion-mmal (either through the web advanced setting or renaming motion-mmal
to motion) had the same effect.  The web interface wasn't active but I could
logon through SSH.  The error in the log file was as indicated in the
thread.  I must have tried at least 10 times in various ways.  Since you
indicated the missing pathway I even tried copying the missing library to
/usr/lib where I thought the other motion libraries where located.

 

Most of the testing was on early release RPi model B.  But I saw the same
effect on my model A running release 6-21 when I switched to motion-mmal.  I
thought the memory split might somehow be an issue.  I tried decreasing GPU
allocation to allow more cpu ram but that didn't work for me.

 

I'd be happy to do more diagnostics.  Alternatively, if could direct me to
patching the pathway to the library I could try that.

 

By the way, I'm interested in your Fast Network Camera pre-release and am
planning to take a look at that as soon as I can.  I really like MotionPie
but I miss some of the camera controls, particularly for low light exposure
of the IR cam.

 

Thanks,

 

Rich Leyden

 

<https://github.com/ccrisan/motionpie/issues/144#issuecomment-113964778> .
<https://github.com/notifications/beacon/AK2mINqYWkGiNJZECfVM2RvDJCWTFDAzks5
oVzDbgaJpZM4Dva9R.gif> 

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

rleyden559, 2015-06-22:
Hello Calin

 

I was trying pre-built images.  I tried release 6-21 and 5-24 before going
all the way back to 3-17.  In each case setting the motion executable to
motion-mmal (either through the web advanced setting or renaming motion-mmal
to motion) had the same effect.  The web interface wasn't active but I could
logon through SSH.  The error in the log file was as indicated in the
thread.  I must have tried at least 10 times in various ways.  Since you
indicated the missing pathway I even tried copying the missing library to
/usr/lib where I thought the other motion libraries where located.

 

Most of the testing was on early release RPi model B.  But I saw the same
effect on my model A running release 6-21 when I switched to motion-mmal.  I
thought the memory split might somehow be an issue.  I tried decreasing GPU
allocation to allow more cpu ram but that didn't work for me.

 

I'd be happy to do more diagnostics.  Alternatively, if could direct me to
patching the pathway to the library I could try that.

 

By the way, I'm interested in your Fast Network Camera pre-release and am
planning to take a look at that as soon as I can.  I really like MotionPie
but miss some of the camera control for long exposures.

 

Thanks,

 

Rich Leyden

 

.Description:
https://github.com/notifications/beacon/AK2mINqYWkGiNJZECfVM2RvDJCWTFDAzks5o

https://www.avast.com/antivirus

ccrisan, 2015-06-22:
@rleyden559, could you please send me the motion.log file right after you reproduce the problem? Does it still complain about a missing library? Please try it using 20150621.

The GPU memory should be set to 128 MB. The fast network camera mode will allow you to control many PI camera parameters.

rleyden559, 2015-06-22:
The GPU was set to 128 MB as default.

I just reset my RPi model A to motion-mmal while it was running release
20150601 (the newest version I've tried).  Unlike my earlier experience I
can longer access it by ssh and it doesn't respond to pings.  I not sure if
that's related but it's very inconvenient for me to physically pull the SD
card at the moment.  Further debugging will be delayed a few days.

 

<https://github.com/ccrisan/motionpie/issues/144#issuecomment-114024150> .
<https://github.com/notifications/beacon/AK2mIMwN7j1DI-Lhlhl2HXDcEI6RZKnuks5
oV6tFgaJpZM4Dva9R.gif> 

---
https://www.avast.com/antivirus

rleyden559, 2015-06-22:
On my Rpi model B (256 MB) I updated fro 20150601 (GPU 128 MB,/usr/bin/motion) to 20150621 by the web update without problem.  Image seems normal.  I deleted old log files. then;

Enabled Fast Network Camera.  After reboot, there is no image (black IE, no camera icon in Chrome).  I tried changing camera settings but nothing I tried yielded an image.

Attached in motioneye.log
>>>>
2015-06-22 15:05:58: INFO: cleanup started
2015-06-22 15:05:58: INFO: wsswitch started
2015-06-22 15:05:58: INFO: thumbnailer started
2015-06-22 15:05:59: INFO: server started
2015-06-22 15:06:00: ERROR: HTTP 400: Bad Request (unknown operation)
2015-06-22 15:06:00: WARNING: 400 GET /picture/1/current/?seq=1435010809598&_username=admin&_signature=dd578f1a609a9d0eb4329bce20c3012a96c1de2c (72.67.31.236) 42.19ms
2015-06-22 15:06:14: ERROR: HTTP 400: Bad Request (unknown operation)
2015-06-22 15:06:14: WARNING: 400 GET /picture/1/current/?seq=1435010810649&_username=admin&_signature=2ee7312e5723a8d9e6e10b507cdb0a6333123e37 (72.67.31.236) 56.07ms
2015-06-22 15:08:53: ERROR: [Errno 2] No such file or directory: '/var/log/motion.log'
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1346, in _execute
  File "/programs/motioneye/src/handlers.py", line 121, in wrapper
    return func(self, *args, **kwargs)
  File "/programs/motioneye/src/handlers.py", line 1407, in get
    with open(path) as f:
IOError: [Errno 2] No such file or directory: '/var/log/motion.log'
2015-06-22 15:08:53: ERROR: 500 GET /log/motion/?_username=admin&_signature=59e76950ea614e744ff30344c11035152b3d4352 (72.67.31.236) 67.72ms
2015-06-22 15:08:59: ERROR: [Errno 2] No such file or directory: '/var/log/motion.log'
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1346, in _execute
  File "/programs/motioneye/src/handlers.py", line 121, in wrapper
    return func(self, *args, **kwargs)
  File "/programs/motioneye/src/handlers.py", line 1407, in get
    with open(path) as f:
IOError: [Errno 2] No such file or directory: '/var/log/motion.log'
2015-06-22 15:08:59: ERROR: 500 GET /log/motion/?_username=admin&_signature=59e76950ea614e744ff30344c11035152b3d4352 (72.67.31.236) 50.04ms
2015-06-22 15:09:16: ERROR: [Errno 2] No such file or directory: '/var/log/motion.log'
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1346, in _execute
  File "/programs/motioneye/src/handlers.py", line 121, in wrapper
    return func(self, *args, **kwargs)
  File "/programs/motioneye/src/handlers.py", line 1407, in get
    with open(path) as f:
IOError: [Errno 2] No such file or directory: '/var/log/motion.log'
2015-06-22 15:09:16: ERROR: 500 GET /log/motion/?_username=admin&_signature=59e76950ea614e744ff30344c11035152b3d4352 (72.67.31.236) 51.27ms
2015-06-22 15:09:41: ERROR: [Errno 2] No such file or directory: '/var/log/motion.log'
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1346, in _execute
  File "/programs/motioneye/src/handlers.py", line 121, in wrapper
    return func(self, *args, **kwargs)
  File "/programs/motioneye/src/handlers.py", line 1407, in get
    with open(path) as f:
IOError: [Errno 2] No such file or directory: '/var/log/motion.log'
2015-06-22 15:09:41: ERROR: 500 GET /log/motion/?_username=admin&_signature=59e76950ea614e744ff30344c11035152b3d4352 (72.67.31.236) 53.64ms
2015-06-22 15:09:47: ERROR: [Errno 2] No such file or directory: '/var/log/motion.log'
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1346, in _execute
  File "/programs/motioneye/src/handlers.py", line 121, in wrapper
    return func(self, *args, **kwargs)
  File "/programs/motioneye/src/handlers.py", line 1407, in get
    with open(path) as f:
IOError: [Errno 2] No such file or directory: '/var/log/motion.log'
2015-06-22 15:09:47: ERROR: 500 GET /log/motion/?_username=admin&_signature=59e76950ea614e744ff30344c11035152b3d4352 (72.67.31.236) 51.90ms
2015-06-22 15:10:28: ERROR: [Errno 2] No such file or directory: '/var/log/motion.log'
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1346, in _execute
  File "/programs/motioneye/src/handlers.py", line 121, in wrapper
    return func(self, *args, **kwargs)
  File "/programs/motioneye/src/handlers.py", line 1407, in get
    with open(path) as f:
IOError: [Errno 2] No such file or directory: '/var/log/motion.log'
2015-06-22 15:10:28: ERROR: 500 GET /log/motion/?_username=admin&_signature=59e76950ea614e744ff30344c11035152b3d4352 (72.67.31.236) 48.33ms
>>>>>>>>>>>>
other log files seem normal
from dmesg.log
>>>>
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
                   vmalloc : 0xc8800000 - 0xff000000   ( 872 MB)
                   lowmem  : 0xc0000000 - 0xc8000000   ( 128 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0xc0008000 - 0xc04c3f0c   (4848 kB)
                     .init : 0xc04c4000 - 0xc04e4000   ( 128 kB)
                     .data : 0xc04e4000 - 0xc0516580   ( 202 kB)
                      .bss : 0xc0516580 - 0xc0568824   ( 329 kB)
>>>>>

Sorry, I seem to be a bad test case.

rleyden559, 2015-06-22:
Still no image in Fast Network Camera mode.  I tried the following over ssh;
..........
root@mp-4e18e722 ~]# python /usr/bin/raspimjpeg.py -w 680 -h 480 -r 2
2015-06-22 16:02:35:  INFO: raspimjpeg.py 0.2
2015-06-22 16:02:35:  INFO: hello!
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
Traceback (most recent call last):
  File "/usr/bin/raspimjpeg.py", line 248, in <module>
    init_camera()
  File "/usr/bin/raspimjpeg.py", line 197, in init_camera
    camera = picamera.PiCamera(resolution=(options.width, options.height))
  File "/usr/lib/python2.7/site-packages/picamera-1.10-py2.7.egg/picamera/camera.py", line 488, in __init__
  File "/usr/lib/python2.7/site-packages/picamera-1.10-py2.7.egg/picamera/camera.py", line 620, in _init_camera
  File "/usr/lib/python2.7/site-packages/picamera-1.10-py2.7.egg/picamera/exc.py", line 191, in mmal_check
picamera.exc.PiCameraMMALError: Camera component couldn't be enabled: Out of resources (other than memory)
...
What other resources, other than memory, are relevant?

ccrisan, 2015-06-23:
This error usually appears when the camera is already in use by another process. @rleyden559 is your problem related to motion-mmal or to Fast Network Camera? Mind you that they cannot (and should not) be used in the same time.

rleyden559, 2015-06-23:
Good point.  I didn't stop any of the Fast Camera processes so there was
already an instance of  raspimjpeg.py running.  Here are the first 3 lines
of "top".

.

592     1 root     S    81444  65%   9% {raspimjpeg.py} /usr/bin/python
/usr/bin/raspimjpeg.py --awb auto --brightness 50 --contrast 0 --drc off
--ev 0 --exposure aut
  610     1 root     S    18372  15%   1% /usr/bin/python2.7
/programs/motioneye/motioneye.py --port 85 --motion-binary /usr/bin/motion
--log-level info --mjpg-client-i

..

Still no image in the web interface and no jpg files being generated as far
as I can determine.

 

<https://github.com/ccrisan/motionpie/issues/144#issuecomment-114432412> .
<https://github.com/notifications/beacon/AK2mIGK3Y-I0qM4BlTHPliPy2nFsDxerks5
oWSregaJpZM4Dva9R.gif> 

---
https://www.avast.com/antivirus

ccrisan, 2015-06-23:
Alright, then let's agree this has nothing to do with motion-mmal. What browser are you using? Did you try to clear the cache?

rleyden559, 2015-06-23:
I agree, it has nothing to do with motion-mmal.  I tried both binaries
/usr/bin/motion and /usr/bin/motion-mmal both "work" the same in release
20150621 for me.  The web interface come up as normal but neither displays
an image.

 

I tried 3 different browsers, Iexploer, chrome, and firefox.  Chrome and
firefox display the camera with the slash icon whereas Iexplorer displays
the camera name on a black background.  Clicking the full screen button
behaves somewhat differently, chrome shows the "no camera" icon immediately,
whereas firefox and iexplorer show the loading icon (shutter) endlessly.
Second PC the same.

Try http://pigcam.icv99.net:85/

 

<https://github.com/ccrisan/motionpie/issues/144#issuecomment-114499296> .
<https://github.com/notifications/beacon/AK2mIHzEodFyhSjsFD5J9AtsY22NJDqLks5
oWVVCgaJpZM4Dva9R.gif> 

---
https://www.avast.com/antivirus

ccrisan, 2015-06-23:
Are you accessing the camera from a public IP address? If yes, try accessing it from your local network. The Fast Network Camera relies on the fact that the stream is pure MJPEG (as opposed to JPEG by JPEG, refreshed by Javascript), for the sake of performance. You need to make the port 8081 accessible (forwarded to your PI if that's the case).

rleyden559, 2015-06-23:
Yes, I'm outside the local network at the moment.  

OK, port 8081 works.  It took me a few tries to get your precise meaning
however.  Setting port  8081 in the expert settings didn't seem to work
correctly.  However setting the web page port to 80 and having the router
forward both 80 and 8081 to the camera IP works OK.  Using [url]:8081 gives
the stream directly in Firefox.

I can look more closely when I get back local.  Perhaps a single port would
work too.

 

Are you still interested in any errors regarding motion-mmal in older
releases should I find them?  Seems moot now since release 20150621 works
for me now with both motion and motion-mmal.

 

Thanks for your patience,

 

Rich

 

<https://github.com/ccrisan/motionpie/issues/144#issuecomment-114524239> .
<https://github.com/notifications/beacon/AK2mICSQWXNTFJsiE59XYOyn-dcyFINhks5
oWWSlgaJpZM4Dva9R.gif> 

---
https://www.avast.com/antivirus

ccrisan, 2015-06-24:
@rleyden559 "Setting port  8081 in the expert settings didn't seem to work
correctly." - Expert Settings port allows you to change the web UI port, not the streaming port. If you set the web UI port to 8081 it will conflict with the existing streaming port which defaults to 8081. That's why it did not work.

Using a single port is impossible as there are two different processes that need to communicate (the web UI and the streamer) and you need access to both of them from the browser, if you use Fast Network Camera.

Yes, I am interested in motion-mmal-related errors as long as motion-mmal is used in a *legitimate* way (i.e. with one RPi camera, a correctly connected ribbon cable, a minimum of 128 MB for the GPU and without Fast Network Camera enabled).

closed 2015-06-24 by ccrisan