ccrisan/motionpie #147: SMB fails on update

tsrgit, 2015-03-18:
MotionPie was succesfully saving video files to an SMB share on my NAS in release 20150112.  Updating to release 20150215 broke it.  In motioneye.log, there is a "failed to mount smb share" error.  The camera won't turn on because of the network share error, and I cannot get it to mount the share.  I tried a clean install of release 20150215 and 20150228, but neither works.  When I reverted to a clean install of release 20150112, I could use my network share again with no errors.  Did something change between release 20150112 and 20150215 that might be preventing me from mounting a network share in MotionPie?

ccrisan, 2015-03-18:
I took a look over all the commits between 20150112 and 20150215. I couldn't find anything suspicious. Can you send me the log files (/var/log/motioneye.log and /var/log/messages) so I can try to identify the issue?

ccrisan, 2015-03-31:
Can you please test the latest version and see if the problem is gone?

tsrgit, 2015-04-01:
It works!  Thank you.

closed 2015-04-01 by ccrisan