ccrisan/motionpie #148: cannot connect web interface with admin account after changing port

joramos21, 2015-03-18:
Hi,

I just changed the port I set (16531) to 8081 and now I cannot connect via the web interface.
It ask me to enter a username and password but it's not the usual motionpie interface that show but the browsers one.
I can connect with the admin account throught ssh.
And I also can connect with the user account.

I tried this because I was able to connect to the motionpie web interface via a 3G connect via a web browser but it didn't work with 4 android apps I tried. I thought that maybe it was due to the port, but I don't think it's the reason.

Thank you for your help.

Best Regards,

ps. the software is very nice, congratulations

joramos21, 2015-03-19:
Hi,

I think I found the problem. I set the same port to the streaming and http port.
I flashed again the image, so don't take time to answer may question.

Thank you :-)

ccrisan, 2015-03-19:
Yes, that could pose some problems :) I'm glad you sorted it out.

closed 2015-03-19 by ccrisan