ccrisan/motionpie #149: cannot connect to motionpie from Android device

joramos21, 2015-03-19:
Hi,

I'm trying since hours to connect from my Android device. I tried 2 different apps.
Tried through the stil pictures (http://192.168.1.9:8081/static/img/camera-progress.gif) and via the video streaming port. The only thing I'm able to see is a black picture with a small one in white color.
I also tried without setting a password to the user account.

Port trigering is ok as I can connect to the web interface via a 3G connection.

Android apps tested : 
_ tinyCam Monitor Pro
_ Net eye camera

Thank you for your help.

José

ccrisan, 2015-03-19:
Try the streaming url (http://192.168.1.9:8081). That's the only one that's supposed to work. Use your surveillance credentials, if requested.

joramos21, 2015-03-24:
Hi once again,

I've read an another thread where you explain de authentification diggest, so that it should work without setting a password to the surveillance user account.
Are you going to do something so that it works with a password ?
I tried your suggesstion but it doesn't work.

Thank you. Have a nice day.

ccrisan, 2015-03-24:
It has to work if you *do not set a password for the surveillance account* and you *do not specify a username or password in your 3rd party applications*. I am planning on adding support for a more common HTTP authentication method (known as "basic HTTP authentication"). That has more chances to work with various 3rd party applications.

joramos21, 2015-03-24:
Thank you very much. It's the only feature that is missing in your great software.

ccrisan, 2015-03-31:
The latest version will allow you to choose basic authentication (or simply disable it) for the mjpeg streaming. This should solve your issue.

closed 2015-03-31 by ccrisan

joramos21, 2015-04-01:
Hi,
Thank you very much. I clicked on the "Check" button from the web interface to try to update but I'm already running latest version. Does it mean that the release you're talking about is not yet available for download ?
Thank you.

fgarciasa, 2015-04-02:
Hi joramos21
Look at the notes of the current version (work fine)

Automatic Update Note
==================

The automatic update mechanism will probably not work due to a SSL certificate checking issue. If you want to use the update function, you'll need to manually patch a file on your sd card:

    log in using SSH or putty (see shell access)
    remount the filesystem read-write: mount -o remount,rw /
    edit the problematic file: nano /programs/motioneye/src/update.py
    add the following line at the beginning of the file: import ssl; ssl._create_default_https_context = ssl._create_unverified_context
    save and exit: Ctrl+O, Enter, Ctrl+X
    run the reboot command
    when your motionPie gets back, the update function should work just fine