ccrisan/motionpie #152: Streaming Authentication broken

lastfreenick, 2015-03-20:
I've just updated my machines to the latest stable and from what I understand, I now need to authenticate if I want to watch the stream on port 8081 and the credentials should be the ones grom the user account. I can log into the webinterface just fine with the user credentials I've set up but if I enter said credentials when prompted for the stream on 8081, the connection is refused. Am I getting something wrong?

ccrisan, 2015-03-20:
The connection on 8081 should never be refused, unless you have disabled the camera streaming for the first camera. In any case, the connection refused error has nothing to do with the credentials.

lastfreenick, 2015-03-20:
Ok I guess, connection refused is incorrect as a technical term. I just can't connect. The authentification promt just keeps popping back up ( with vlc player on linux) if I try it via the browser (chrome) I get the same behavior. Used to work with 0215 build.

ccrisan, 2015-03-20:
Ah, well in that case it might be related to the authentication. First off, if you want to try it in a browser, use Firefox. Did you set a password for the surveillance user? Did you change the surveillance username from `user` to something else?

lastfreenick, 2015-03-20:
The surveillance name is user I just changed the password. Authentification indeed works on firefox but fails on vlc and the webcam app I use to monitor the streams on my wall mount home automation remote android tablet thingy.

ccrisan, 2015-03-20:
In that case those apps might not support HTTP digest authentication. I may be able to add an option for choosing basic HTTP authentication instead of the digest one.

lastfreenick, 2015-03-21:
As a workaround I've disabled user password for now. I'll gladly try out for you in case you get around to support HTTP basic authetication. 

And as always, thanks for your outstanding support!

snagglewest, 2015-03-22:
Some camera apps/programs such as zoneminder ect will accept a

 camera name of - username:password@your.ip.address.xxx
 and  stream url as - http:your.stream.ip.adress:8081 format

Try various combinations until you figure out what you're app wants

karma007, 2015-03-23:
hello,

authentication not working with google chrome on Windows 7 (and cannot add the remote cam in motionpie using streaming url), does not work with IE11, does not work on iOS (ipad/iphone).

Not working on VLS (iOS and Windows)

Removing the user password works

thanks

ccrisan, 2015-03-31:
The latest version allows you to choose the authentication method to be used by the mjpg video stream.

closed 2015-03-31 by ccrisan