ccrisan/motionpie #153: red boxing

piecam, 2015-03-20:
HI ccrisan,

This is not really an issue, but a question. Why is it red boxing sometimes and not sometimes?
thanks
[image]
[image]

dkelley810, 2015-03-20:
That usually indicates movement was detected.

dkelley810, 2015-03-20:
I have a question for you? Do you actually have four cameras? and are they plugged into one Pi? Or multiple Pi's?

piecam, 2015-03-20:
HI dkelley810,
Two pis with pi Cam and usb cam on each pie.

thanks

dkelley810, 2015-03-21:
Did my explanation of the red box help?

closed 2015-03-24 by ccrisan