ccrisan/motionpie #157: Firewall

thegnnu, 2015-03-24:
Hi Calin Not a issue just a question due to my stupid Virgin Media SuperHub stopping me getting into the motionPie camera's from the web even had port forwarding active the only way was to turn on DMZ so as all ports are now open to the raspberry pi. Would I be able to install a firewall onto the Pi via SSH without causing problem to your software.
TerryR

ccrisan, 2015-03-25:
Just use the `iptables` command available in motionPie. The following lines would do the trick:

    iptables -P INPUT DROP
    iptables -A INPUT -p tcp --dport 80 -j ACCEPT
    iptables -A INPUT -p tcp --dport 8081 -j ACCEPT
    iptables -A INPUT -p tcp --dport 22 -j ACCEPT

There's no support for custom commands to be run at startup (i.e. no `rc.local`), but I will consider adding one.

thegnnu, 2015-03-25:
Thanks for your reply I will try that on my return home

TerryR

ccrisan, 2015-03-31:
The latest version has support for user-defined [startup commands](https://github.com/ccrisan/motionPie/wiki/Tweaks#adding-custom-startup-commands). You could place your firewall rules in there.

closed 2015-03-31 by ccrisan