ccrisan/motionpie #159: TPlink wireless not working with Pi2

jbauer3102, 2015-03-27:
I have a B+ model that I use a TPLink nano wireless adapter with and it works great.  I purchased a Raspberry Pi 2 and I have tried every version you have posted for the Pi2 and I can not get the wireless to work.  I configure the settings the same.  I still have the wired network connected at this point and when I click to apply the wireless changes it requires a reboot.  After a re-boot when the nano adapter is plugged in the webpage can not be displayed.  Again the wired connection is still there and I am tying to view the page from the wired IP address.  The wireless connection never shows up in my DHCP client table.  I remove the wireless adapter and the webpage comes up fine.

This is the same set up I used on the B+ so I know it should work. I am wondering if there is a hardware difference with the Pi2 that does not like the TPLink wireless adapter.

ccrisan, 2015-03-27:
There is definitely no significant difference between the two versions. Some logs would he helpful. Can you connect a monitor and (take a photo to) see what it says when it comes to initializing the wireless connection? Also make sure the power supply is good enough.

jbauer3102, 2015-03-30:
Sorry for the delay getting back to you. Had a busy weekend.  

After connecting the monitor I can see that the unit gets caught in a boot loop for some reason. It gets to the part where it says starting motioneye, it pauses for a few seconds and then a few lines come up before it says requesting reboot now.
 I have attached some pictures I took with my phone.  Hopefully this will help.  If you need me to get any log files or anything let me know.

[image]
[image]

ccrisan, 2015-03-31:
Thanks for the screenshots. Clearly the firmware for your driver is somehow missing. I'll make sure to include it with the next release.

jbauer3102, 2015-03-31:
Thanks!

BTW Awesome product!!  

I am configuring a whole home security system for less that $200.  Which is amazing, if you get that donate button going, I am in.

jbauer3102, 2015-03-31:
Found the donate button.  Sent a few bucks your way for an awesome product.

ccrisan, 2015-03-31:
Please try the latest [20150331](https://github.com/ccrisan/motionPie/releases/tag/20150331) release and should the problem persist, just report back the issue.

closed 2015-03-31 by ccrisan