ccrisan/motionpie #160: Multiple camera recording

dandymon, 2015-03-27:
The newer version doesn't allow the two cameras to record.
Whilst both cameras appear in the menu, and both allow you to edit the settings (etc..), only one camera's recordings appear in the list.
It appears to be the first USB port that the camera is added to motion pi.
I've switched cameras around the USB ports, and it still favours the first camera's USB, even though they are different cameras.
I've tried to remove both and re-add them, reload the software, but it won't record two sets of video any more. I've checked the SMB path and there are no second camera files - it's not just a case of only showing one camera's listings

Thanks
Dan

ccrisan, 2015-03-28:
Please attach the /var/log/motion.log and /var/log/motioneye.log files. I would also like to see the configuration files. Remove any sensitive information before sending them.

dandymon, 2015-03-28:
Hello Chris, please find attached.

Sorry I don't have Linux so the config is in a zip file - so as I could
remove the password data. - let me know if you would rather just the conf
files decompressed.

thanks
Daniel

dandymon, 2015-03-28:
I think I might have found the error.

The second camera appeared to freeze after a short amount of time, thus not
allowing any recordings.
I put the frame sizes down lower and reduced the FPS, and it seems to be ok
for now - but it never had any issues on the older version... (it's fair to
say it might be my hardware).

Still, being able to over-clock the rpi2 will probably alleviate this error.
Do you think this is possible? (currently the overclock feature is not for
the RPi1 and causes it to crash).

ccrisan, 2015-04-12:
You can test the overclocking capabilities of your RPI2 yourself. Each piece is different and will have a different overclocking "crash" limit.

closed 2015-04-12 by ccrisan