ccrisan/motionpie #161: Add: Upload to Online Hoster

fnordson, 2015-03-28:
Thank you very much for that. I had a pretty broken motion construct running. I was very happy to ditch the whole monster and replace it with your beautiful system.

However. I have one problem. I need to watch a room where there are no running computers I could copy the output to. The main problem in this room is, that the computers are what gets stolen. I'm pretty sure they would take the PI also if they find it.

I need a way to upload the recorded files to Google Drive, Drop Box or similar so the videos are safe. Could you implement such a feature? I've seen a way to do it here: http://raspberrywebserver.com/serveradmin/back-up-your-pi-to-your-google-drive.html but I have no idea how to build that into your system and I don't want to break it (since I'm not that much of a crack in this).

ccrisan, 2015-03-28:
I have plans to implement such a feature, but i currently don't have too much time to take care of it.

fnordson, 2015-03-28:
Thank you for answering. 
Do you see any possible ways for me to implement the solution above without breaking mopie?

Odin73, 2015-03-29:
Until ccrisan has time to implement this feature, i had my PC at home running Blue Iris cctv software then the 2 raspberry pi cams installed at work running motion pi connected over the net to my home pc, caught one employee I suspected of stealing red handed. hope this helps

adriangibbons, 2015-03-29:
I run a bash script stored in /data (where the photos are stored. You'll need to connect via SSH to your RaspPi and create the script as shown in the screenshots attached.
```bash
#!/bin/bash
now="$(date + '%Y-%m-%d')"
find $now -type f -exec curl --user username@server.com:password --app
rm -rf $now
```
[image]
[image]
[image]
[image]
[image]

fnordson, 2015-03-29:
Thank you adriangibbons. Did not think about that "run a command" option. Good idea. I'll try to get it running through that.

adriangibbons, 2015-03-29:
I've just updated to the latest release and see that you can now connect via FTP. So another option would be to create a webhook and have a script connect via FTP to pull the images; rather than push as my method above.

fnordson, 2015-03-29:
OK that sounds good. I don't understand it but it sounds good. 
I'm off googling "webhooks"  :D

matteopiccioni, 2015-04-08:
+1 request for upload the recorded files to Google Drive, Drop Box or similar
Thanks for your work!

Volyova, 2015-06-22:
+1 on this. I would really love to set up several of those in our kindergardens since the break in rate rose so high, the insurence won't finance any more professional alarm equipment. Not that the one before would have helped...
I would love to hide some PIs around the rooms where computers stand and even if they find and steal them (like the rest of the equipment) to have the data securely saved.

Please make this happen Mr. Crisan. 
I'm not very good at the stuff and the solutions above look cryptic to me :/

Volyova, 2015-08-10:
Any chance for an ETA on the feature Mr. Crisan?

ccrisan, 2015-08-10:
Nope, no ETA yet :)

Volyova, 2015-08-10:
Alright. 
We're watching you ;)

ccrisan, 2015-08-10:
I'm rather busy with netcam issues and improvements these days. Uploading to remote servers will come soon though :)

AlRiley, 2015-08-20:
Excellent work on the MotionPie Calin.  I've installed it as a webcam with the idea of uploading the photos to Weather Underground.  I've had look at Adriangibbon's idea but cannot see where the command is for directing the upload to is at.  Any ideas or shall I wait in case you are thinking about implementing a feature that can do this soon?

martymoose, 2015-08-21:
I ill put my hand up as well
thats a feature i need as well
ftp upload to weather underground for my weather station
Still great work

*Martin Thompson*

*Po Box 1415*
*AYR   QLD    4807*

*M. 0466 005866*
*E.  marty@martymandy.com
<https://mail.google.com/mail/u/0/?view=cm&fs=1&tf=1&to=marty@martymandy.com>*
*W. **martymandy.com* <http://martymandy.com>

Check the weather at my place martymandy.com

fawcett, 2015-08-25:
+1 on functionality to push videos to other hosts.

ccrisan, 2015-08-25:
It's one of the first priorities :)

JakobRieckh, 2015-09-26:
Thanks Crisan for your great work! 

Same here: Would be great to be able to push pictures directly to Dropbox & Co.

malcolmct, 2015-10-15:
I just added instructions to synchronize with Google Drive at https://github.com/ccrisan/motioneyeos/issues/4