ccrisan/motionpie #162: Raspberry Pi 2 Camera module, not available

dewnut, 2015-03-28:
Hi -I have installed MotionPie - Everything boots up OK, and the web page is accessible, but the following messages indicate that the Pi CSI camera is not set up correctly:
You have not configured any camera yet. Click here to add one...
An error occurred. Refreshing is recommended.

When I try to run 'Add Camera', the only options offered are 'Network camera' and 'Remote motionEye camera' - there is no option for the Pi camera.
Raspberry Pi 2
motionpie-raspberrypi2-20150317
OV5647 camera. 

Thanks!

dewnut, 2015-03-28:
Thanks, no I have a RP2

ccrisan, 2015-03-31:
Please check the [latest version](https://github.com/ccrisan/motionPie/releases/tag/20150331). Your camera might be among those for which support was added recently.

dewnut, 2015-04-01:
Thanks!  I will try that.  I actually have plugged in a USB webcam for now
and got that to work!

I have a real stupid question for you... I cannot seem to view a streaming
shot from : http:myipaddress:8081 the still images work fine though...
http://myipaddress/picture/1/current/

Is there something I need to do additional to get streaming to work?

ccrisan, 2015-04-01:
Don't try that from Chrome. Chrome doesn't like mjpegs opened directly in a browser's window. If you must see that in a browser, try Firefox.

dewnut, 2015-04-01:
ah - gotcha, thanks!  I was using Chrome.  IE 11 doesn't work either.
Firefox works great!

Now, to view it outside my network, I just need to forward port 8081on my
router for myipaddress?

ccrisan, 2015-04-01:
IE has no idea what mjpeg is. Yes, you'd normally want to forward 80 and 8081.

dewnut, 2015-04-01:
huh. weird.  Can't seem to get that to work... [image: Inline image 1]

dewnut, 2015-04-02:
I figured it out... I was trying to connect to http://externalIP:8081 ...
if I just do http://externalIP I am able to see the stream.

Thanks again for the help!

ccrisan, 2015-04-02:
Alright! No problem.

closed 2015-04-02 by ccrisan

dewnut, 2015-04-02:
do you have any suggestions on how to view the stream on an iPad?

dewnut, 2015-04-02:
haha never mind.  works fine in the native browser!  :)

dewnut, 2015-04-03:
sorry to keep bothering you.  it appears that my images/video are not
saving, is there something I am missing with the setup?

ccrisan, 2015-04-03:
How did you configure the File Storage section? What about Motion Detection and Motion Movies. A screenshot that shows all of them would help me identify your issue.

dewnut, 2015-04-03:
I got it figured out.  I think it was an issue with my SD card.  Ended up
rebuilding it with the newest release which is working great!  Thanks
again!  Great work!