ccrisan/motionpie #164: wpa_supplicant.conf - PSK Special Characters - Infinite Reboot

Taeous, 2015-03-30:
I've been playing with attaching my MotionPie (RaspberryPi 1 B+) to my wireless network. I have several other cameras on this network, and use a 63-character long PSK that incorporates letters, numbers, and special characters (but no spaces). After editing my /data/etc/wpa_supplicant.conf file and rebooting, it hangs for a moment (and ultimately fails) on 'Starting wpa_supplicant'. It will continue to boot (also failing on setting the current data via http). It will start motioneye, and then error with:

[57.809081] bcm2835-v412: error 0 waiting for frame completion
The system is going down NOW!
[58.204550] rtw_cmd_thread(wlan0) _rtw_down_sema(&pcmdpriv->cmd_queue_sema) return _FAIL, break
Sent SIGTERM to all processes

At this point it will infinitely reboot.

Configuring a PSK with non-special characters works fine (although I haven't yet tested which characters are breaking things). I can tell you that the special characters I use in my secure PSK are: `+_,?;'-*#!

On a side note, is there any way to implement encryption of the PSK in the config file? I'm not a big fan of putting my passwords in plain text.

ccrisan, 2015-03-30:
If editing the /data/etc/wpa_supplicant.conf file did not work for you, then I don't know how to fix your problem. You should look for a solution on other Linux forums and maybe keep the special characters at a minimum. In my opinion they don't improve the security that much.

Regarding encrypting the file, I don't think that is possible; at least not without an enormous amount of work to implement an encrypted folder (using eCryptfs), using mount namespaces and so on.

CryssDOTnet, 2015-04-25:
Why don't special characters improve security that much?

ccrisan, 2015-04-26:
I won't start a debate on this topic here, but I'll say that each character added to the key makes the number of required test combinations 128 times bigger (roughly), while using a non-alpha-numeric character instead of an alpha-numeric one wouldn't make any difference to a brute-force attack.

Of course we're assuming that no dictionary words are used when setting the password so that a dictionary attack is out of question.

Taeous, 2015-04-26:
Let me start off by saying that I agree with ccrisan - I just have a few clarifications I'd like to make.

The 'strength' of a password depends on many factors, such as how it's stored, if the hash value is salted or not, etc. - not just which character sets are used or how long it is.

If you're only looking at brute-forcing a wireless password, then, as ccrisan stated, the length of the password is far more important than the complexity of the password (side note: it CAN be dictionary words strung together - these have no bearing on how secure a password is...as long as it's (seemingly) random words and nothing predictable like a sentence). XKCD summed it up pretty well: https://xkcd.com/936/

This does not mean that there is no value in using special characters though. When attempting to brute-force a password, you have to try different character sets: A-Z, a-z, 0-9, standard special characters (!@#$%,. etc) (there are more, but these are the standard ones). If all things are equal (the same length, hashing, etc), a password that combines upper, lower, and number character sets will take less time to crack than one that combines upper, lower, number, and special characters. This is simply because there are more possible combinations for the given length of the password when you add another character set into the mix.

Again - length of a password is FAR more important than if it uses special characters or not. But, there is still value in using special characters if possible.

closed 2015-05-16 by ccrisan