ccrisan/motionpie #165: Extra motion options is striping ampersand from url

galactic-void, 2015-03-30:
I'm trying to setup an ip camera with the user/password embedded in the url however each time the config for the camera is saved the ampersand in the url is striped.

Example (Extra motion settings):
`netcam_url http://<ipaddress>/videostream.cgi?user=name&pwd=pass`

Result:
`netcam_url http://<ipaddress>/videostream.cgi?user=namepwd=pass`

Expected:
`netcam_url http://<ipaddress>/videostream.cgi?user=name&pwd=pass`

ccrisan, 2015-03-30:
It's fixed in upstream motionEye with the latest commits. The next motionPie version will hopefully work just fine for you.

ccrisan, 2015-03-31:
The [latest version](https://github.com/ccrisan/motionPie/releases/tag/20150331) should fix your problem.

closed 2015-03-31 by ccrisan