ccrisan/motionpie #166: Login credentials reset

abocin, 2015-03-31:
I'm not able to pass the Login screen.

The Motionpie install went fine. I was able to see some live images from the camera. However, while looking for the setting button (which wasn't located on my Motionpie landing page) I've hit the security button. While introducing some credentials I wasn't able to pass by the Login display. I haven't introduced previously any user and password.

How can I pass the Login screen or reset the credentials in order to get access?

Thanks.

abocin, 2015-03-31:
OK. I got it. admin

ccrisan, 2015-03-31:
Alright!

closed 2015-03-31 by ccrisan

abocin, 2015-03-31:
I have though one additional question.

The camera seems to have some latency. Is there any chance to reduce the latency below 1s?

Thanks.