ccrisan/motionpie #168: Sticky keys (physical keyboard)

Cooper-Dale, 2015-03-31:
Known issue with bad / old driver ... please update it. ref http://elinux.org/R-Pi_Troubleshooting#Sticky_keys

ccrisan, 2015-04-01:
The firmware as well as the kernel are as up-to-date as they can be.

fah, 2015-04-05:
I tried 3 keyboards (Microsoft, Dell, Apple) all showed:
"error opening USB device 'descriptors' file

None of the 3 keyboards worked.

Pi2B, Version 20150331

ccrisan, 2015-04-12:
There seems to be a problem indeed with keyboards and RPI2. The weird thing is that none of the "solutions" found online makes sense for this problem. The firmware as well as the kernel are as up-to-date as they can be. The power supply that I use is good. I still experience the sticky keys and erratic behaviour with my simple Dell keyboard.

Any help regarding this issue will be appreciated.

anjoze, 2015-04-23:
I have the same problem with RPi2 and Sticky keys.
I've made some tests:
Rpi2 + MotionPi + 3 different keyboards = Sticky keys
Rpi2 + Raspbian + 3 different keyboards = No Sticky keys
Rpi1 + MotionPi + 3 different keyboards = No Sticky keys
Rpi1 + Raspbian + 3 different keyboards = No Sticky keys
For all the tests the Power unit was the same: 5V 1200mA

ccrisan, 2015-04-23:
Some recent commits to the RPI firmware as well as to the RPI kernel seem to do something about this issue. I'll make sure the next motionPie version uses the latest commits and let's hope for the best.

ccrisan, 2015-05-16:
I still haven't been able to find the cause of this issue. I use the [latest Raspberry PI kernel] (https://github.com/raspberrypi/linux) and the issue is still there. Any suggestion is appreciated.

MSumulong, 2015-06-30:
I believe this is an eudev issue: https://github.com/gentoo/eudev/issues/110, which I think should be resolved in the next buildroot release.

ccrisan, 2015-06-30:
Does it still happen with the latest release? I remember fixing it a month or two ago.

MSumulong, 2015-07-01:
Looking at your master branch, the package/eudev/eudev.mk shows it is using v3.0. The issue says it will be fixed for v3.1. The latest buildroot (http://git.buildroot.net/buildroot/tree/package/eudev/eudev.mk) shows its been upgraded to v3.1.1.

ccrisan, 2015-07-01:
We're not talking about the same issue. The original problem was caused by a misconfiguration of the kernel, only for the RPi2. If the problem still persistsz it may have another cause (eudev) now.

ccrisan, 2015-07-01:
So does the problem still happen? The eudev issue is in fact just an error message that appears in the kernel log, without affecting any functionality?

MSumulong, 2015-07-02:
I'm not sure. I haven't been able to recreate the problem. I only have an RPi1 on hand.

closed 2015-07-02 by ccrisan

anjoze, 2015-07-03:
Hi ccrisan,
I cant test it now because I dont have my RPi with me but when possible I'll tell you if it's solved or not.
Thank you