ccrisan/motionpie #169: unable to open video device

abocin, 2015-04-01:
I'm using motionpie on Raspberry Pi B+ with Raspberry Pi camera. 
Installed motionpie-raspberrypi-20150317
Wi-Fi 300Mbps

Using Ethernet I could connect to camera
Using WiFi I'm getting "unable to open video device"
[image]

abocin, 2015-04-01:
The motion.log says:

[ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "netcam_http"
[0] [NTC] [ALL] motion_startup: Motion trunkREV561 Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (EMG)

ccrisan, 2015-04-01:
Use the latesr version (20150331). Manually install it from scratch. It fixes your issue.

closed 2015-04-01 by ccrisan

abocin, 2015-04-01:
Thank you. It worked Ethernet and WiFi. Also, the camera was disconnected physically from the socket. :)