ccrisan/motionpie #170: Mask File integration

lastfreenick, 2015-04-01:
Hi Calin,

great update again, thanks for fixing streamig authentification!

Another thing that I've been wanting to try out is Mask Files:
http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigOptionMaskFile

First of all, do you think this will make my poor pi sweat even more in the performance department?

Secondly, I guess adding an web based editor for Mask Files will be a major pain in the behind, so are are there any chances for a more basic inclusion, maybe an option in expert settings to enable mask files or maybe even an upload dialouge for a mask file I edited myself?

In case this is possible, I'll be happy to add a wiki section that expains how to generate the pgm file externally (e.g. on the users computer).

As always thanks a bunch and keep up the great work!

Colin from Germany

ccrisan, 2015-04-01:
The idea is to implement a user-friendly editor right in the web UI. Until then, you can simply add the extra motion options manually.

I'm not sure about the performance as I never tested this feature before.

jan-r, 2015-04-08:
I'm using a mask file with motionPie on my Raspi 2, and it doesn't seem to affect the streaming performance. It does, however, prevent my poor Pi from having to dump several thousands of still images to the SD card which result from moving trees and leaves, so it really improves FTP performance when reviewing images.

jimmyeao, 2016-03-13:
Has anyone done a 'how to' on this? It would be really useful :)