ccrisan/motionpie #172: Pi password not updated by motionPie

jepolch, 2015-04-04:
Pi 2 model B. I'm using the Pi 2 distro. When I change the administrator password using the web interface it works for the web login. However, the admin password is not working for root in SSH. I can use the serial number as the root password, but not the admin password. When I was using motionPie on an older model B, the admin password could be used to log into the Pi as root. I wonder if this is related, from /var/log/messages:

Apr  4 09:24:40 mp-xxxxxxxx auth.info sshd[1094]: WARNING: /data/etc/moduli does not exist, using fixed modulus

ccrisan, 2015-04-04:
Do you run the latest version? Did you reboot after changing the password? Did it ask for a reboot?

jepolch, 2015-04-04:
I'm running 20150317. It did offer to and I did reboot after the password change. When I SSH as root using the new password, the response is "Access denied". Here are the log entries for the failed (using UI password) and successful (using serial) logins:

Apr  4 12:49:53 mp-75ccxxxx auth.info sshd[19738]: WARNING: /data/etc/moduli does not exist, using fixed modulus
Apr  4 12:49:59 mp-75ccxxxx auth.info sshd[19738]: Failed password for root from 192.168.1.151 port 60568 ssh2
Apr  4 12:51:33 mp-75ccxxxx auth.info sshd[19738]: Accepted password for root from 192.168.1.151 port 60568 ssh2

ccrisan, 2015-04-04:
If you're able to connect a monitor, there's a short message saying something like "setting default root password: 75ccxxxx" or "setting custom root password". Which one of those appears? Does it appear at all? Is it followed by an immediate error?

jepolch, 2015-04-04:
"Setting custom root/admin password" I didn't see any errors.

Thanks very much for looking into this and thanks again for your great program.

ccrisan, 2015-04-04:
Actually it's not the first time someone reports this exact issue. I'd really like to track it down and see what's really happening. Can you please log in remotely (using SSH) and try to manually set the password to your admin/UI password: `passwd <your_pwd>`? Then see if that works. There might be some special characters or so that are not accepted for some reason.

jepolch, 2015-04-04:
I had tried that before and it failed. I just did it again to capture the error message:

[root@mp-75ccxxxx ~]# passwd
Changing password for root
New password:
Retype password:
passwd: can't create '/data/etc/shadow+': File exists
passwd: can't update password file /etc/shadow
[root@mp-75ccxxxx ~]#

jepolch, 2015-04-04:
You might want to see this:

[root@mp-75ccxxxx etc]# ls -l shadow*
-rw-------    1 root     root            53 Dec 31  1969 shadow
-rw-------    1 root     root            55 Dec 31  1969 shadow+
-rw-------    1 root     root            54 Dec 31  1969 shadow-

jepolch, 2015-04-04:
I renamed shadow+ and shadow- and tried to set the password again and got this:

passwd: /etc/passwd: Read-only file system
passwd: can't update password file /etc/passwd

[root@mp-75ccxxxx etc]# ls -l /etc/passwd
-rw-r--r--    1 root     root           559 Mar 15 13:43 /etc/passwd

jepolch, 2015-04-04:
Aha! Success. After I renamed shadow+ and shadow- I changed the password in the web UI and it rebooted. Then I tried the new password in SSH and it worked. Just to be sure, I changed the password again in the web UI and tried it in SSH and it worked the second time as well. And now in /data/etc there is no shadow+.

[root@mp-75ccxxxx etc]# ls -l shadow*
-rw-------    1 root     root            53 Dec 31  1969 shadow
-rw-------    1 root     root            54 Dec 31  1969 shadow-

ccrisan, 2015-04-05:
Thanks for your time. Actually it was the `shadow+` file that was causing the problems. The rest of the errors can be safely ignored.  Somehow `shadow+` was kept there from a previous boot or so. The next version will make sure to remove it before attempting to set a password.

closed 2015-04-05 by ccrisan