ccrisan/motionpie #173: ssh terminal fails

fah, 2015-04-05:
When I connect from a current OS X via ssh I get:
`no matching cipher found: client aes128-cbc,3des-cbc,blowfish-cbc,cast128-cbc,arcfour,aes192-cbc,aes256-cbc server aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com,chacha20-poly1305@openssh.com`

ccrisan, 2015-04-05:
Apparently you need to enable these ciphers in your client's configuration file. [this](http://mgalgs.github.io/2014/10/22/enable-arcfour-and-other-fast-ciphers-on-recent-versions-of-openssh.html) is what I found online regarding your issue. Let me know if this helps.

fah, 2015-04-05:
Yes, you are right. Since that never happened before I guessed thats on your side. Sorry!
**Solution:**
In the error message you have to read the word "server" and the following values. These are the ciphers possible on server side.
I picked "aes256-crt".
Now change the line (Mac OS X 10.10.2) in */etc/ssh_config* or *~/.ssh/config* 
`Ciphers aes128-cbc,3des-cbc,blowfish-cbc,cast128-cbc,arcfour,aes192-cbc,aes256-cbc`
to 
`Ciphers aes128-cbc,3des-cbc,blowfish-cbc,cast128-cbc,arcfour,aes192-cbc,aes256-cbc,aes256-ctr`

closed 2015-04-05 by fah