ccrisan/motionpie #174: Cannot add second remote motioneye camera

kbmetz, 2015-04-06:
I've tried adding a second MotionPie camera and get the same failure no matter what I do. I select 'add camera', then 'Remote Motion Eye Camera' and give it the URL. It fails when I use a password and when I don't (I try changing the remote camera config with and without a password). Each time the error is the same "Operation timed out after 10001 milliseconds with 1631139 bytes received".  Screen shot added
[image]

ccrisan, 2015-04-06:
If you typed in 8081, you're probably trying to add a MJPEG camera (or the MJPEG streaming part of a motionEye/motionPie camera), also known as *network camera* or *IP camera*. motionEye cameras usually run on port 80 or 8765.

The error you're seeing is due to the fact that a MJPEG stream is an *infinite* request sending JPEGs, and thus the URL tester will never receive a response.

Depending on what you actually want to do, either change the Device to Network Camera or type in the correct URL to a motionEye instance (that must otherwise load in a new browser window if you try it).

kbmetz, 2015-04-06:
So, I've tried every other URL and it always says 'camera not found'. For example http://192.168.50.104:8765/, or http://192.168.50.104:8765/picture/1/current/
What is the default 'correct' URL to a default motionEye instance? If I just type in the IP and port as shown above I get 'not found' so I'm a bit stumped at the moment. Thanks

ccrisan, 2015-04-06:
You're right, there seems to be a bug. Thanks for reporting it.

kbmetz, 2015-04-06:
Great. if I can help troubleshoot or anything please let me know

bleuchien, 2015-04-08:
It seems to be commit **[16ec614]** that leads to double slashes in the url which is not recognized on the server side and leads to a 404:
```
2015-04-08 15:44:36: ERROR: HTTP 404: Not Found (not found)
2015-04-08 15:44:36: WARNING: 404 GET //config/list/?_username=admin&_admin=true&_signature=8309b289051c6102e1652bfb12fff7ba9146a2c5 (10.25.7.30) 38.73ms
```
If I remove the ```/``` from the variable ```uri``` on the client side (```main.js```) during debugging it works. However I've not checked what else breaks if the commit is reverted.

[16ec614]: https://bitbucket.org/ccrisan/motioneye/diff/static/js/main.js?diff2=16ec614d8ea0&at=master

ccrisan, 2015-04-08:
Thanks for taking the time to identify the cause of the bug. I haven't had too much time on my side to take a look over this problem.

Reverting the commit would break adding netcams running motion daemon, netcams which require a trailing slash in the url to work.

I'll find a better way to fix this.

kbmetz, 2015-04-09:
I can confirm the same. I edit the main.js, removed the '/', and I was able to add a remote motionEye using http://192.168.1.5 (note no trailing backslash). So this particular bit works for me as well but I don't know what else it would break either. I'll update this issue if I find it breaks anything

bleuchien, 2015-04-09:
It may not be the nicest solution but you could remove all double slashes on the server side in *remote.py (line 28)* inside ```def _make_request(...)```  by simply substituting doubles with singles:
```python
import re

def _make_request(host, port, username, password, uri, method='GET', data=None, query=None, timeout=None):
    uri = re.sub('//+', '/', uri)
    url = '%(scheme)s://%(host)s%(port)s%(uri)s' % {
            'scheme': 'http',
            'host': host,
            'port': ':' + str(port) if port else '',
            'uri': uri or ''}
...
```

ccrisan, 2015-04-12:
Fixed with [this](https://bitbucket.org/ccrisan/motioneye/commits/1b1aae2348f0a47a94ccc310715237193f884f79?at=master) commit.

closed 2015-04-12 by ccrisan