ccrisan/motionpie #177: Pi 2 and framerate

jtengland, 2015-04-09:
I am using motionPie with a RasPi 2 and a Pi Cam and I am seeing strange behavior with the video files that are captured. It does appear there are some issues with framerate, but I haven't found a thread that really hashes it out.

At 800x480 I get files that claim to be 29-30fps, at 1024x768 I get files that claim to be 19fps, and 1280x720 nets me files that claim to be 17fps.

BUT, when I try to play them back, the first few seconds either appear to play normally or at a sped up framerate, and then the remaining 28 seconds of the files play at what appears to be 1fps... Another interesting thing is that the file size (3-5.5 MB) seems consistent with what I'd expect from a normal 17-30fps framerate.

Is there anything I need to tweak to fix this? Is this codec related? I know Motion has had reported framerate issues, but I assumed there was some hope since there's a framerate slider in the web GUI. :D

Thanks!

ccrisan, 2015-04-09:
Try reducing the framerate. Even if th PI2 is capable of streaming at such a high pace, it probably isn't capable to encode them. The result is probably something like what you get.

jtengland, 2015-04-09:
Thanks for the fast reply. Reducing the framerate produces identical results. 

I've seen smooth video from a Pi, so I know it can be done somehow. Is this specifically a limitation of Motion?

ccrisan, 2015-04-09:
It most likely is. Try reducing the framerate even more. I run my cameras at 640x480@5fps with PI1 and everything seems to be just fine. PI2 should come with some performance improvements but not too many.

closed 2015-04-12 by ccrisan

TechGiga, 2015-09-21:
I tried running the camera at 480p at 5 fps along with the network stream but still streamed at 1fps, which is identical to when I set it to 24fps. Any ideas?

ccrisan, 2015-09-22:
There are two framerates: the video device framerate (at which images are captured from device) and the streaming framerate (at which the images are streamed). Make sure both of them are set to 5fps.

TechGiga, 2015-09-22:
Yep, they are both set at that.

ccrisan, 2015-09-22:
Can I see the config files (motion.conf, thread-x.conf) and the logs (motioneye.log, motion.log)? Make sure to remove sensitive data.

TechGiga, 2015-09-22:
Where are they located? (I have SFTP access)

ccrisan, 2015-09-23:
Well, the files are at `/data/etc` and `/var/log` respectively. You could however download them directly from the UI: the configs come with the backup archive and the logs can be individually downloaded from the Expert Settings section.

TechGiga, 2015-09-24:
All the motion.conf file consists of are usernames and passwords and nothing else. Here are the other logs: http://www.gigawertblog.byethost17.com/motionpielogs/

ccrisan, 2015-09-24:
The config file seems alright to me and the logs show nothing suspicious. How can you tell it's 1fps?

TechGiga, 2015-09-24:
Well, the stream moves to the next frame only when the second has changed, or sometimes even less.

arkins, 2015-10-11:
I have the same issue, where I turned off streaming, only leave still & motion picture on.
Base on the property, some avi files are saying 10 fps, but video looks like all 10 frames with same image.

Will file name of still affect video result? for still image, I say capture all frames, but I just found that filename, I only specify up to second.

ccrisan, 2015-10-11:
Don't save all frames as they could slow down the system and it doesn't make too much sense for common setups. Of course two pictures taken within the same second would be saved in the same file, so, with the default filename convention you couldn't save all the frames. And again, please don't :)