ccrisan/motionpie #178: Sirius USB 2.0 camera

jepolch, 2015-04-11:
I have three cameras, one of which is identified by motionPie as a "Sirius USB2.0 Camera". This camera seems a little sensitive to light when using Windows software, but with motionPie, it's pretty much unusable. I thought I had received a bad camera, so I returned it to Amazon and the second one has the same issue. It doesn't even respond to the Brightness setting in the UI. Is there any hope for this cheap camera? Which (cheap) camera works best with motionPie?

It's Camera1

8 PM http://i61.tinypic.com/14nctbs.jpg

11 AM http://i59.tinypic.com/2zogb69.jpg

ccrisan, 2015-04-12:
See how it behaves on a normal Raspbian with any kind of v4l2 utility. motionPie (and motion) will simply  adjust the brightness, contrast and hue controls of the v4l2 device. If that fails, there's not much I can do about it. Maybe some module (driver) parameters?

jepolch, 2015-04-22:
It's no better when using 'avconv -f video4linux2 -i /dev/video0 -r 5 -s 640x480 -f mjpeg -qscale 5 - 2>/dev/null | ./streameye -d' (apparently avconv is the follow-on to ffmpeg) on Wheezy. As I said, it only seems to work fine in Windows. Perhaps it will work better with a later version of motionPie.

ccrisan, 2015-04-23:
Then it's a faulty driver. I'm sorry but you'll need to wait for someone to update the driver. I suggest buying another type of camera.

closed 2015-04-23 by ccrisan