ccrisan/motionpie #179: Config Options Being Overwritten

nexusuk, 2015-04-13:
The wiki mentions...

*"You can make changes to these files as you please, but keep in mind that various options corresponding to the settings found in the UI settings panel will be overwritten when applying settings in the UI. However configuration variables that are not covered by the UI are left untouched when settings are applied."*

I'm finding that the entire file is being reset including options (such as smart mask settings and motion/debug images) that are not included in the UI.

Where should I be including motion config options that will allow them to persist between UI updates and reboots (motioneye.conf maybe?)

Thanks

ccrisan, 2015-04-21:
My apologies for getting back with an answer this late. Normally what the wiki says should be correct. Any additional config options that are present in `/data/etc/thread-x.conf` should be preserved. Moreover, they will appear in the UI, in the *Extra Motion Options* field, under the *Video Device* section.

I believe that you're editing `/data/etc/motion.conf` instead of the `thread-x.conf` file. While it would be a good idea to also preserve additional lines in `motion.conf`, it doesn't work that way right now. In any case, no other options should be added to `motion.conf` but rather to the `thread-x.conf` file specific to your camera, as they are probably options that concern one specific camera.

Please let me know if this is the case and if the solution proposed solves your problem.

nexusuk, 2015-04-28:
Hi, 

Thanks for your reply. I've been trying again to edit /data/etc/thread-1.conf and I'm still running into the same problem. I'm specifically trying to use the 'output_motion on' config option to generate debug images (I'm not sure if this is the correct config option for the version of Motion being used here) however when I try to enable (or disable) the motion detection setting in the UI and apply that change when I then reload thread-1.conf in nano to check the contents the additional config option has been removed

ccrisan, 2015-04-29:
`output_motion` is one of the settings supported by the UI (and is thus overwritten using the setting from the UI). Enable `Motion Movies` if you want `output_motion` to be set in the configuration file.

closed 2015-04-29 by ccrisan