ccrisan/motionpie #182: Wrong Date and Time

Hoenerk, 2015-04-16:
Everything seems to be running fine with my camera except that it is registering the wrong date and time. It seems to think it's New Years 1970, and the time is two hours behind. The minutes are off too, but only by a few.

ccrisan, 2015-04-17:
Make sure your motionPie has access to the internet. Otherwise it won't be able to set the current time.

Hoenerk, 2015-04-17:
The motionpie is wirelessly connected to the network, just like any other wireless internet device. Do you know if there is some setting in router I can check to make sure it has access to the internet?

ccrisan, 2015-04-18:
You may want to connect a monitor and see what it has to say when it tries to set a date.

anjoze, 2015-04-20:
Hi.
I had that problem and I think that's a bug.
I had everything working and connected with ethernet cable.
Someone unpluged the ethernet cable this weekend.
Today the date is 1970. It shouldn't change to that date just because it's offline for some time.

ccrisan, 2015-04-20:
I will try to reproduce it.

ccrisan, 2015-04-21:
I cannot seem to reproduce the problem. Normally there's a script that will make sure to update the system time each 15 minutes. As long as there's internet connection, the date should be alright.

If the connection goes down, depending on the watch settings, your PI may reboot or not. Either way, when the connection gets back, the script should still be able to update the time.

Is this problem reproducible? If yes, does your PI always show 1970 dates or does it lose the date/time later?

Hoenerk, 2015-04-21:
If I reboot the motion pie it goes back to Dec 31, 1969. If I leave it alone it continues to move forward without any issue. Somehow I guess it can't connect to the internet to correct itself.

ccrisan, 2015-04-21:
Have you configured static IP? Do you have an outbound firewall in your network? Does your network require an HTTP proxy setting?

In any case, please connect a monitor to the HDMI output and take a photo of it to see what it has to say when trying to update the system time.

anjoze, 2015-04-23:
Ok, I think I've figured out what was causing this.
I've disconnected the ethernet cable 2 days and the date/time still OK.
Then I though that maybe the electricity was down too. So I powered OFF RPi and turn ON again and the date was 1970.
So my conclusion is: in my case, the ethernet cable was disconnected and electricity was OFF. Then, when electricity came up, the ethernet cable still off and RPi couldn't get the correct date. Because there is no Real Time Clock the date is the beginning.
No problem with software.

ccrisan, 2015-04-23:
Thanks for getting back to me. Alright then, I'll mark the issue as invalid.

closed 2015-04-23 by ccrisan

Hoenerk, 2015-04-23:
I was not using an Ethernet cable when this issue occurred. The MotionPie
is hooked up to the network using WiFi. When it is hooked up with the
Ethernet cable the date is still 1970. I think you were closer when you
suggested it was a firewall issue. I have not had time to look into it, but
you're probably right that it isn't an issue with the software. Anyway, not
a big deal.

P.S. the last messages you sent arrived here at 2:40 AM, in case you were
wondering why I didn't respond.

Hoenerk, 2015-04-23:
You were right in your previous message from 2 days ago, the static IP was
configured incorrectly. The MotionPie was attached to the local network, I
could access it from any computer on the network, but it was not able to
make a connection to the internet. Now the MotionPie reads the correct date
and time. Thank you for taking the time to make suggestions. This really
was a minor problem, and the software you've created is really great. Thank
you for making available, and for the support and time your giving to us.

ccrisan, 2015-04-23:
It's always a pleasure when it works out right :)