ccrisan/motionpie #184: possible helping out RPI CAm Web Interface with your motion/Eye/motionPie

piecam, 2015-04-17:
HI ccrisan,

I was trying to encourage you to give a hand to this following project with your great strength and wonderful motionEye system.
http://elinux.org/RPi-Cam-Web-Interface

I know that you have declined a few times and you didn't have much time and interest in this. I just could not help it but want to encourage you once more.

Currently, their project can do the followings, as I understand,
1. Use MMAL api (not actual /dev/video0) to get 1080p video and 2594 stills.
2. Use motion to monitor a smaller resolution image and trigger 1080p or 2594 recording.
3. They are using web gui to achieve other stuff, like, time setting, scheduler, etc.

I feel that your work motioneEye/motionPIE are so wonderful. If you can help them a bit by incorporating your development to their core functions, it will create a great fusion of system.

It might be something worthwhile that you might give it a try of their system. If you tried their system and you'd like to merge your code with theirs. I would feel that's going to be great!

With your code, we can,
1. use one motion-thread1.conf to manage RPI WEB Interface core function (tap into mmal api, not /dev/video0)
2. use other motion-thread2-xxsx.conf to manage other USB cam
3. have a wonderful interface, instead of "messy" apache/web/css etc.

Opensouce world on RPI Web Cam Interface project is calling for your help!!!Please help!!! :-)

Hope that my being stubborn in persuading you , doesn't not offend in any way. 

Once again, I feel your work is great! if you can reconsider helping them out, it is even more wonderful!

Many thanks

ccrisan, 2015-04-17:
Adapting the motionEye frontend to anything that is not motion (i.e. that doesn't comply with its configuration options and that does not stream mjpegs) is a task that demands a lot of work. I don't have the time for this kind of work, but if anyone else is willing to try to do this work, I'd be more than happy to provide guidance.

On the other hand, I have recently started working on a small add-on for motionPie. Its goal is to turn your PI into a simple and performant MJPEG streamer with minimum fuss and taking as much advantage of the GPU as possible. It will use the RaspiMJPEG tool you mentioned to feed JPEGs to another little tool (still in development) that acts as an HTTP server. Of course the motion daemon is not part of this scenario, so there will be no motion detection or text overlays in this mode.

This way one could use one PI device running motionPie in fast MJPEG mode to simply do streaming and a second PI device (or a more performant computer) to do the motion detection using (ideally) motionEye.

piecam, 2015-04-17:
It's great that you are working on new tools/method utilizing RaspiMJPEG, which is actually core part of that RPI Web Interface project. 

You know much more than I do for these things. However, it looks like that RPI Web Interface, indeed uses motion to monitor a cam feed and trigger some actions. They do motion detection and text overlay. Hope that I am talking the same things as you might intend. And their CPU is not much actually, with detection. It has no a circular buffer so that you can get "pre-motion" portion of image/video as well.

Anyway, just want to pass on some thoughts and hope Pi related secure cam project can get better and better with so many talents, like youself, devoting time and effort and love.

Thanks again.

ccrisan, 2015-06-23:
Part of what this issue describes here is implemented by the new Fast Network Camera feature.

closed 2015-06-23 by ccrisan