ccrisan/motionpie #186: cron

henkth, 2015-04-18:
Dear Calin Crisan,

Congratulations making motionPie! A surveillance camera with Raspberry Pi has become so much more user friendly now!
For many users who are "on the go" somewhere I think the best feature request would be to have an automatic dynamic ip update feature, as most of us are on constantly changing ips automatically provided by our providers. It's therefore not possible to see the webcam now on the net. For the time being the adding of cron would help, so we can tweak motioneye & motion with a dynamic ip update script and a free dns account somewhere. I realise this has already been asked a few times by other users, just pointing out the importance.
Thank you for considering,

Henk Thoma
Leh, Ladakh
INDIA

ccrisan, 2015-05-31:
The latest version has cron support. [Here](https://github.com/ccrisan/motionpie/wiki/Tweaks#running-commands-at-regular-intervals-cron-support)'s a small article on how to use it. I hope this helps.

closed 2015-05-31 by ccrisan

henkth, 2015-05-31:
Dear Calin Crisan,

Absolutely fantastic! Thanks for adding this feature.

Kind regards,

Henk Thoma