ccrisan/motionpie #187: Turn Off Camera LED

henkth, 2015-04-19:
I am using the Raspberry Pi camera module (5MP, 1080p) with motionPie. Problem is that the red LED is on all the time and I want to make this inconspicuous. As far as I understand I can change the line disable_camera_led=0 to disable_camer_led=1 in the config.txt in /boot.
Problem is that it is a read only file and even after trying mount o remount,rw / and chmod 755 config.txt it remains read only. What can be done?

henkth, 2015-04-19:
Oeps, sorry found the answer already under Tweaks. I had to also type:
mount -o remount,rw /boot

closed 2015-04-19 by henkth

ccrisan, 2015-04-19:
If you use a recent version of motionPie you can change that from the UI, in the Expert Settings as well.

henkth, 2015-04-19:
Ah ok. Sorry missed that. Thank you so much for letting me know!