ccrisan/motionpie #190: Loses Wifi

zalsson, 2015-04-27:
Having problem with my two pi running motionpie.
both of them loses the wifi connection in 1min -> 1hour  after start up, and it will not reconnect again.
Some times after reboot/ cutting power they reconnect but most of the time they dont connect.
Works flawless with ethernet cable.

jimmyeao, 2015-04-27:
I can confirm i have seen this issue - in addition, if you plug the ethernet cable in, the wireless interface comes back up.

ccrisan, 2015-04-28:
I would like to know what wireless adapters (maker, model) you use. I would also be interested in the messages log file (downloadable from the Expert Settings UI section, if running the latest version).

Also make sure the power adapter supplies enough current to drive the PI as well as all the peripherals (wifi adapter, cameras, etc).

zalsson, 2015-04-29:
both uses edup wifi adapters

http://www.dx.com/p/ultra-mini-nano-11n-usb-2-0-802-11n-b-g-2-4ghz-150mbps-wifi-wlan-wireless-network-adapter-54193#.VUDfvpOsW94
http://www.dx.com/p/ultra-mini-nano-usb-2-0-802-11n-150mbps-wifi-wlan-wireless-network-adapter-48166#.VUDiU5OsW94

power supply 2700mA and one 12v 3000mA with step down to 5v.

Message log files
https://www.dropbox.com/sh/rf04cshz0lw69eu/AAB-QgFLiuNaFIgFDItIA5Pua?dl=0

ccrisan, 2015-04-29:
@zalsson, from the logs you provided, it seems that your PI suddenly stops. It doesn't look like an issue generated by the wireless adapters.

The new version will provide an option to download the `boot.log` file that contains more details on the boot/init process.

zalsson, 2015-04-29:
Wierd that both have the same problem then...
Any tips ? :)

ccrisan, 2015-04-29:
Well I wouldn't totally exclude the possibility that the wireless adapters cause power issues or simply crash the kernel and the watchdog kicks in. You'll need to try to use a different wifi adapter and see if the problem persists. This way we can rule out or blame the adapters.

zalsson, 2015-04-29:
They runs fine with raspian and kodi.
So should not be hardware issues.
But will try some other adapters

ccrisan, 2015-04-29:
Can you also try to disable overclocking (if you have it enabled) in Expert Settings and see if it makes any difference?

zalsson, 2015-04-29:
Its off :( already

ccrisan, 2015-04-29:
Try connecting a monitor and see what actually happens when the PI goes off like this.

dampy78, 2015-05-04:
Did this get resolved at all? Mine has also started doing the same with an edimax ew-7811. Was working flawlessly for a month until today. If I plug in a Ethernet cable the WiFi works again.

ccrisan, 2015-05-05:
It's very difficult to debug network related issues without detailed boot logging. The new version (yet to be released) will allow you to download a boot.log file that clearly says when the device has been rebooted and why. Please hold on for another few days till I manage to release the new version.

Jasimo, 2015-05-08:
Hi, just tried motionpie the first time yesterday (great work, thx ccrisan for developing this) and also got problems with my edimax wifi adapter. Looking forward for the version with the ability to download the log.
Just one question, is it normal that after setting up the wifi in the GUI (SSID, and Pass) and performing the required reboot, the setting for wifi is gone in the gui?

ccrisan, 2015-05-08:
The wifi settings should be persistent. I'll try to reproduce this issue myself but normally the values should be there in the UI all the time.

Jasimo, 2015-05-08:
When I first boot the motionPi on my Pi2 (after flashing the sdcard) it is connected via LAN, as soon as I can reach the gui (over LAN) i go to the Wifi settings and enter my parameters (ssid, pass). The gui tells me to perform a reboot. After performing the reboot the PI is still connected via LAN, could this be the issue?

ccrisan, 2015-05-08:
Well, motionPie wasn't designed to be used with multiple connections so this could be a problem. I was wondering:
1) Is the issue 100% reproducible (i.e. can you reproduce it each time starting with flashing the image)?
2) If you repeat the test but unplug the ethernet connection when rebooting, does it connect to the WiFi? Does it show the WiFi settings in the UI?

Jasimo, 2015-05-08:
Will try all of that at the weekend and come back to you.

Jasimo, 2015-05-08:
I just tried it again, and it react as follows.
Boot Pi2 with newly flashed motionpi image. Connect via Lan and set Wifi to "on" in the gui and entered all needed things lik ssid, pass. Shutdown the pi and disconnect lan. Boot the pi with a succesfull Wifi connection. Everything works.
But, in the gui the switch for wifi is "off" ;-(

ccrisan, 2015-05-08:
There is indeed a certain combination of config changes + apply where some settings may be lost. At what point do you enable advanced options and do you hit apply right after that?

Jasimo, 2015-05-09:
Hi, just after the first LAN-Boot, I enabled WiFi, entered my ssid/pass and klicked apply. After that i do the recommened reboot without the lan connected to the pi.
In the meantime i got really problems with different wifi adapters and motionpi, when i switch back to lan everything is stable. I loose my wifi connection every now and then, this makes me crazy.
I know that the edimax adapters got a powersaving function, with raspbian i know how to disable that but i don´t know with BuildRoot. Could that be the reason for the wifi problems?

ccrisan, 2015-05-09:
The powersave function is disabled for all the common adapters, including the Edimax 7811un model. I have done a lot of work for the next version (yet to be released). Let's hope things will become more stable then.

Jasimo, 2015-05-09:
I am sure that the next version is improved. I am just testing with my Edimax and Fritz!Box and it looks like that WiFi Channel 12/13 is not compatible with the Edimax Adapter, maybe this is the problem in my case and for others. Will make some further tests and come back

Jasimo, 2015-05-10:
nothing to do with the Channel it also happens on Channel 6. 
I installed raspbian on by Pi2 and have the same problem on raspbian, an iwconfig shows me that power Managemen is "off" for wlan0, but it insn´t. 
I have to 
sudo nano /etc/modprobe.d/8192cu.conf
and enter a line like this:
options 8192cu rtw_power_mgnt=0 rtw_enusbss=0

after that the power management is really "off" and i got a stable Wifi connection for over 6 hours now.
Could it be that something like this also happens under BuildRoot?

ccrisan, 2015-05-10:
The options line for the 8192cu module is present by default in motionPie. I doubt that it's a problem of power saving. I've been using Edimax 7811un adapters myself with motionPie and I must admit I experience connection problems every now and then. That's why motionPie has all those watchdogs, to be able to detect connection losses and reboot. If your PI does not reboot upon WiFi connection losses, there might be a bug or an uncovered case with these watches.

Please hold on until I manage to release the new version and then, if the problem persists, you'll be able to easily download and attach all the necessary logs.

Jasimo, 2015-05-10:
Will wait for the new Version. In the meantime I don´t think that it is a wifi problem anymore, if i look in the message log it looks like that during the wifi disconnect the Pi is booting.

ccrisan, 2015-05-10:
Most likely it detects a connection problem and triggers a reboot (there's no better way to handle it). The new version will provide a boot.log file that will show all these problems.

Jasimo, 2015-05-17:
Installed the new preview version and leave it running over night (connected via Wifi). In the morning i cannot reach the web-gui and it is not possible to ping the Pi2. Some Minutes later i got a ping response and the gui is reachable. Is there a way to send you the logs?
rgs
Jan

ccrisan, 2015-05-17:
Hi @Jasimo. I took a look over your logs and it seems that your PI rebooted at Sat May 16 20:33:32 UTC 2015 (I don't know if that's really your morning or not). The reboot was not preceded by a clean shut down so I assume the power was cut somehow (is your power supply alright?).

Other than that, there's no sign of connection loss or rebooting due to detectable problems.

Jasimo, 2015-05-17:
Thx for your support. "Sat May 16 20:33:32 UTC" is the time when I disconnect Lan and switched to Wifi, since then the Pi2 runs without restart, fine!
But sometimes, I lose the ping to the Pie and then the GUI is not reachable. Some time later (10 up to 120 sec) it came back again. Do you have an idea what could be the reason for this?

ccrisan, 2015-05-18:
I must admit I myself have experienced ping problems (losses) with Raspberry PIs and Edimax 7811Un adapters. I never managed to identify the underlying problem and I thought it was an isolated case (unfortunate combination of PI, WiFi adapter and possibly WiFi router).

I tried to further investigate the issue and found out that even the arping command was failing at receiving an arp reply. The ping would start working shortly within a minute or two, just like you mentioned.

All the power management disabling hints that can be found online have already been applied to the motionPie OS. I really don't know what else to do about this issue, other than suggesting to enable the Connectivity Watch option (in Expert Settings). This will regularly try to establish a TCP connection with a given host and upon failure will reboot the unit. It could also double as a keepalive mechanism, in case a lazy WiFi adapter would detect silence and would go to sleep.

Jasimo, 2015-05-18:
good to be not alone with the ping loss ;-) Important is that the Pi2 is recording when motion is detected and I think he is doing it very well with the latest preview version
Could you explain the meaning of "WARNING: control hue not found for device /dev/video0" in the motioneye.log

ccrisan, 2015-05-18:
Just ignore the warning. The v4l2 driver does not expose a control for hue - no biggie.

Jasimo, 2015-05-18:
Some minutes ago I was again not able to reach the web-gui for over 5 Minutes (also no ping), but MotionPie is tracking motion and write to my nas during this period.
Is it something insite BuildRoot which cause the problem (maybe Busybox)?

ccrisan, 2015-05-18:
I think not. I experienced the same issue with Raspbian. I believe it's related to either the adapter or the USB which is far from perfect on a raspberry pi.

stezani, 2015-05-23:
Hi, i have the Edimax wifi dongle and i red this link:
http://itsbrent.net/2014/05/fixing-raspberry-pi-wifi-dropping/
could help you?

ccrisan, 2015-05-23:
Nope. Those workarounds have already been applied in motionPie.

Jasimo, 2015-05-26:
I am now on motioneye (Raspbian) for a couple of days and got no ping timeouts or unavailabilty of the gui there. I am using the same hardware as for motionpie.

ccrisan, 2015-05-26:
@Jasimo are you on PI2? Have you tried motionPie on PI1? Does it exhibit the same behaviour? I don't have a PI2 in use with motionPie so I couldn't tell for sure that PI2 works well on wireless. It may be a problem that affects USB devices on PI2, just like the keyboard problem that is a known issue and for which I haven't found a fix yet.

Jasimo, 2015-05-26:
no sorry, didn´t have a pi1 to test it ;-( but i could borrow one of my colleagues at work and give it a try.

Jasimo, 2015-05-28:
No luck also with the Pi1 which shows the same behaviour. My tests are all based on Edimax Wifi Sticks, I wanna now try diffrenet ones.
@ccrisan is it possible to get a list of Wifi adapter which will work out of the box with MotionEye?

ccrisan, 2015-05-28:
@Jasimo, I believe it's the router + adapter combination that matters so it's impossible and useless to maintain a list of adapters that work out-of-the-box. I have been using Edimax 7811Un devices for many years now with Asus routers and I find the combination *usable*.

Now there are many many reported problems when it comes to RPi USB connection and I think that could also be a factor. If the issue is so easily reproducible on your side can you please try varying different conditions a bit and see if the problem goes away? For example:

* try using a totally different power supply (that could have less ripple/voltage fluctuations).
* try placing your PI close to the router and see if the problem diminishes at least
* if you're able to change your router settings, try changing the encryption type

Jasimo, 2015-05-29:
@ccrisan thx for coming back to this topic.
I am using a AVM Fritz!Box 7490 with WPA2 (CCMP) encyption and I am now thinking that the edimax adapter has problems with this. I got ping times between 50ms up to 800ms and of course "no answere" messages in between the continuous ping.
Over the weekend i will try:
* a different Wifi Stick (Logilink WL0084B v.2.0)
* different Power Supply (Apple Ipad Charger)
* Pi2 direktly beside the Fritz!Box
* changing the encyption from CCMP to TKIP (WPA2 to WPA)

Will come back with the results to this topic.
rgs
Jan

Jasimo, 2015-05-30:
Yesterday evening I changed the Edimax Wifi Stick to a Logilink WL0084B (RT5370 Ralink Chipset instead of Realtek Chipset) and all my problems are gone. Ping times between 2ms and 6ms and stable.
This looks like that I defenetly have Problems to bring my AVM Fritz!Box 7490 together with Realtek Chipsetbased Wifi Sticks (tried three different ones).
I also tried a different PSU, without luck. Didnt try to change the Wifi encription to WPA jet.

ccrisan, 2015-05-30:
Thanks for the update. I recently installed smokeping on a local computer to be able to watch over the PIs and see how their connectivity varies with adapters, time and overclocking.

One of the things i noticed so far (3+ days) is that one of them has < 1% ping loss, while the rest of them are constantly reachable.

All of them have been using the edimax adapter. I changed the adapter on the faulty one (to some ralink-based chip) and the problem persists. I will try more adapters and will (hopefully) eventually find the cause to this problem.

Jasimo, 2015-06-01:
I am quite sure now that **my** "Wifi loss problem" can be solved by using a ralink-based chip.
I also got the drops under Raspbian.
For me it looks like that my Fritz!Box 7490 WPA2 (CCMP) WiFi with a 63 character key is to much for the Realtek chipset.

zalsson, 2015-10-08:
Any solution yet?

rushless, 2016-03-11:
Hi,

This topic is in-line with the issue I am having.

First, I love the software thank you for producing it! I am using it to monitor my 3D printer. It might be overkill but MotionPie was so easy to setup I tried it. 

Gear:
RasPi v1B+
Edimax WiFi dongle

I was able to install the os and get it running. I was able to access it via IP address, but walked away for an hour. When I came back, the page had frozen and won't load on refresh. 

I then unplugged my Pi from power hoping to do a reboot. I'm still not able to access the GUI through the device's IP anymore.

Any suggestions? It's set up watching a print but after I'm going to take it down and check the files on my PC, but it was working once!!!