ccrisan/motionpie #191: Refreshing is reccomended

jimmyeao, 2015-04-27:
This has reared its head again today, I can connect to the pi, but there is no video (a camera with a cross through it) and  when applying any settings I get the 'Refreshing is recommended' Reboots don't appear to fix this, and I have tried from a couple of computers / browsers. Clearing my history has no effect.

jimmyeao, 2015-04-27:
Interesting.
If I mount the filessystem as rw and reboot it starts working again..

ccrisan, 2015-04-28:
Please send me the motion.log, motioneye.log and messages log files (downloadable via the Expert Settings UI section). You can email them to me.

jimmyeao, 2015-04-28:
Emailed, thanks

ccrisan, 2015-04-29:
@jimmyeao, your motioneye.log file is full of `ERROR: failed to start motion: [Errno 12] Cannot allocate memory`. This indicates... well... pretty much what it says. If you're using an older 256 MB model, I recommend buying a 512 MB one (at least).

If you're already on a 512 MB PI, the only advice I can think of is to reduce the memory usage as much as possible by disabling various services (such as Samba, SSH, FTP), reducing the memory allocated to the CSI camera (but not too much though) or even reducing the resolution.

closed 2015-04-29 by ccrisan

jimmyeao, 2015-05-19:
@ccrisan This is still occuring - I have made sure the GPU memory is at 128 (It is the 512mb version)
Motion Log is empty, motioneye.log doesn't download.
If I mount the filessystem as rw (mount -o remount,rw /) and reboot it starts working again..

ccrisan, 2015-05-22:
What do you mean motioneye.log doesn't download? If you're able to log in remotely and remount the system r/w you could also take a look at /data/log/motioneye.log and see if you still find the Cannot allocate memory errors (mind the timestamp at the beginning of each line).

If the problem is 100% reproducible, would you be willing to set up SSH remote access to your PI so I can log in remotely and try to identify the issue? A similar issue has been reported and I believe there might be a problem indeed.

jimmyeao, 2015-05-22:
If I can get it to reproduce, Ill set up an SSH session, no problem..I
meant clicking on the link to download the log file did nothing, it
may have been there, but it wouldnt download through Chrome. 

----- Original Message -----
From: "ccrisan/motionpie" 
To:"ccrisan/motionpie" 
Cc:"Jimmy White" 
Sent:Fri, 22 May 2015 01:57:41 -0700
Subject:Re: [motionpie] Refreshing is reccomended (#191)

	What do you mean motioneye.log doesn't download? If you're able to
log in remotely and remount the system r/w you could also take a look
at /data/log/motioneye.log and see if you still find the Cannot
allocate memory errors (mind the timestamp at the beginning of each
line). 

	If the problem is 100% reproducible, would you be willing to set up
SSH remote access to your PI so I can log in remotely and try to
identify the issue? A similar issue has been reported and I believe
there might be a problem indeed. 

	—

Links:
------
[1]
https://github.com/ccrisan/motionpie/issues/191#issuecomment-104580040