ccrisan/motionpie #193: Webcam going offline

guadacorte, 2015-05-02:
Hi Guys
Running on a raspberry pi 2.
I have 2 usb webcams connected. Occasionly one of them will go offline. How can I bring it back online without deleting the camera and setting it back up again?
Thanks

ccrisan, 2015-05-02:
What do you mean by "going offline"? There might be an issue with one of them (a driver bug? a power-related problem?); in any case, you don't need to delete the camera and readd it. Does a reboot bring everything back online?

Please send me the logs (downloadable from the Expert Settings UI section) so I can try to identify the root cause of your problem.

guadacorte, 2015-05-02:
Thanks for your reply. Rebooting does not bring back the camera.
I will look at exporting the logs.

ccrisan, 2015-05-02:
If a reboot does not bring back your camera, then I suspect a hardware problem (possibly in combination with a faulty driver).

guadacorte, 2015-05-02:
I'm going to stick with one camera for a while.

ccrisan, 2015-05-02:
Well you could always try another (type/model of) camera, as long as your hardware performs well enough.

dpatonsr, 2015-05-09:
I have a similar issue. Twice now my main pi camera has locked up.  No image.  The log file says something about not being able to write. 
I have two cameras.  My second camera is not affected.  I can connect to it separately just fine.  When i log into the first camera.  No video is shown.  If I try to restore the config,  it fails with the message to "refresh recommended". The first time this happened i just reformatted the sd card and started over.  I would prefer not to do this once a week. .. 
I have a NAS attached and have been accessing the first camera thru port forwarding on my Apple router.  I've sent the log files, please let me know if there is anything i can do to fix it before i reformat again. Thanks!

guadacorte, 2015-05-09:
Mine now locks up if I add another camera. I always shut down the pi before adding a camera. After powering up and adding a camera It becomes disconnected from the network within 10 minutes. I have tried using wireless and cable. Still the same problem. I thought it was a power supply issue so I bought a 5.25 volt 2 amp one from modmypi. Still having the problem.

dpatonsr, 2015-05-09:
Mine works fine for about a week.  Both times it had locked up,  I first
noticed the problem when accessing the camera from a low bandwidth remote
location. Not sure if that matters.  I wondered if the data is not being
pulled fast enough from the pi and that creates the problem somehow.  I
have the frame rate at 5, with the 30 fps and the 1600 x 1200 frame size.
Thanks

ccrisan, 2015-05-09:
@dpatonsr if you're on a PI1, either 5 fps or 30 fps at 1600x1200 is a little too much for the CPU. You may want to decrease the size or switch to Pi2.

dpatonsr, 2015-05-09:
Hi Calin,
I should have mentioned that I am using the pi2.

ccrisan, 2015-05-10:
@dpatonsr if the log files say something about not being able to write and you're using a NAS, I suspect a connectivity problem with that NAS. The motion daemon displays the "no video" image when it's not able to write the generated files to the underlying filesystem.

ccrisan, 2015-05-10:
@dpatonsr There are many "OSError: [Errno 12] Cannot allocate memory" messages in your motioneye.log file. If you're really on a PI2 (1GB RAM), then there must be a process that eats up a lot of memory (or maybe a memory leaker?).

If you are able to remotely log into your PI whenever your problem occurs, you could run the following command to identify the process that eats that much memory:

    ps -o rss,vsz,comm,args

I would like to see the output of this command.

borlandoflorida, 2015-05-19:
Hi,

I am also experiencing a similar issue. I have MotionPie running on my Raspberry Pi B+ with two Microsoft Xbox Live Vision cameras connected via USB. I am networked using ethernet because the power became an issue with two cameras and a wifi dongle connected via USB.

Everything has been running smoothly for a few days - I could access both cameras and motion stills / videos are being stored on my NAS drive without issue.

However, this morning when I logged in and checked the cameras, there were two grey squares instead of the live feeds. After a little tinkering, I found out that I could now only manage one camera instead of two (if I unplugged one camera, the other would show and vice versa).

I have just checked the logs and I have the following error in my motion log:

    ERROR: failed to relay event: [Errno socket error] [Errno 111] Connection refused

I'm also seeing the following messages in the messages log:

    auth.err passwd: can't create '/data/etc/shadow+': File exists
    auth.err passwd: can't update password file /etc/shadow

And finally, in my motioneye log, I've started to see the following message:

    OSError: [Errno 12] Cannot allocate memory
    2015-05-18 23:00:07: ERROR: Exception in callback <functools.partial object at 0xb6455810>
    Traceback (most recent call last):
    File "home/ccrisan/pi/motionPie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/ioloop.py", line 565, in _run_callback
    File "home/ccrisan/pi/motionPie/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/stack_context.py", line 275, in null_wrapper
    File "/programs/motioneye/src/thumbnailer.py", line 71, in _run_process
    _process.start()
    File "/usr/lib/python2.7/multiprocessing/process.py", line 130, in start
    File "/usr/lib/python2.7/multiprocessing/forking.py", line 121, in __init__

I also have added the streaming URL to an addon to Kodi (OpenELEC on my Pi 2) called multicam, so I can open this up and see the stream on my TV and I suspect that since adding this, the problem has started. The reason I suspect this is because I see this message in my motioneye log:

    2015-05-17 14:09:45: ERROR: 500 POST /config/0/set/?_=1431868186013&_username=admin&_signature=***MASKED*** (IP.***.*.***) 175.42ms

I'm going to try again tonight without the addon in Kodi running to see if that fixes anything, but would love any feedback!

ccrisan, 2015-05-19:
Thanks for your detailed report. The first error (the one with the event relaying) indicates the motionEye server was down at that moment. It's probably a consequence of the same out-of-memory problem.

The passwd-related errors can and should be ignored. They always show up and have nothing to do with your problem.

The final error clearly indicates a memory issue. One of the processes is leaking memory. Just like I asked @dpatonsr before, I'm asking you to run the following command on an SSH remote shell on your PI when this problem occurs:

    ps -o rss,vsz,comm,args

It will show us what process eats most of the memory.

borlandoflorida, 2015-05-19:
Ok, thanks. I will have a look at some point (maybe tonight depending on time) and report back with the output from the command you provided. 

Would the last issue I posted have anything to do with the memory by any chance, or is that what you are hoping to identify after running the command over SSH?

This is the issue I am referring to:

    2015-05-17 14:09:45: ERROR: 500 POST /config/0/set/?_=1431868186013&_username=admin&_signature=***MASKED*** (IP.***.*.***) 175.42ms

ccrisan, 2015-05-19:
Nope, that's just an error indicating the request failed. It probably failed due to the same memory-related problems.

borlandoflorida, 2015-05-19:
Ok, thanks. Will let update this when I have tried what you have suggested.

borlandoflorida, 2015-05-19:
Ok, so I have just run the command you suggested and this is what I have:

RSS  VSZ  COMMAND          COMMAND
 572  968 init             init
   0    0 kthreadd         [kthreadd]
   0    0 ksoftirqd/0      [ksoftirqd/0]
   0    0 kworker/0:0      [kworker/0:0]
   0    0 kworker/0:0H     [kworker/0:0H]
   0    0 kworker/u2:0     [kworker/u2:0]
   0    0 rcu_preempt      [rcu_preempt]
   0    0 rcu_sched        [rcu_sched]
   0    0 rcu_bh           [rcu_bh]
   0    0 khelper          [khelper]
   0    0 kdevtmpfs        [kdevtmpfs]
   0    0 khungtaskd       [khungtaskd]
   0    0 writeback        [writeback]
   0    0 crypto           [crypto]
   0    0 bioset           [bioset]
   0    0 kblockd          [kblockd]
   0    0 kworker/0:1      [kworker/0:1]
   0    0 cfg80211         [cfg80211]
   0    0 kswapd0          [kswapd0]
   0    0 fsnotify_mark    [fsnotify_mark]
   0    0 cifsiod          [cifsiod]
   0    0 VCHIQ-0          [VCHIQ-0]
   0    0 VCHIQr-0         [VCHIQr-0]
   0    0 VCHIQs-0         [VCHIQs-0]
   0    0 dwc_otg          [dwc_otg]
   0    0 DWC Notificatio  [DWC Notificatio]
   0    0 kworker/u2:1     [kworker/u2:1]
   0    0 deferwq          [deferwq]
   0    0 kworker/u2:2     [kworker/u2:2]
   0    0 mmcqd/0          [mmcqd/0]
   0    0 kworker/0:1H     [kworker/0:1H]
   0    0 ext4-rsv-conver  [ext4-rsv-conver]
   0    0 kworker/0:2      [kworker/0:2]
   0    0 jbd2/mmcblk0p3-  [jbd2/mmcblk0p3-]
   0    0 ext4-rsv-conver  [ext4-rsv-conver]
   0    0 VCHIQka-0        [VCHIQka-0]
 596  952 syslogd          /sbin/syslogd -n
 520  960 watchdog         watchdog -t 5 /dev/watchdog
1160 2000 udevd            /sbin/udevd -d
1448 1924 udevd            /sbin/udevd -d
1436 1912 udevd            /sbin/udevd -d
1248 1892 udevd            /sbin/udevd -d
1392 1892 udevd            /sbin/udevd -d
1404 1912 udevd            /sbin/udevd -d
1396 1896 udevd            /sbin/udevd -d
1256 1940 udevd            /sbin/udevd -d
1264 1948 udevd            /sbin/udevd -d
1272 1952 udevd            /sbin/udevd -d
1276 1956 udevd            /sbin/udevd -d
   0    0 kworker/0:3      [kworker/0:3]
 824 1136 dhclient         dhclient -cf /var/cache/dhclient.conf eth0
1104 1920 S40network       {S40network} /bin/sh /etc/init.d/S40network start
 980 1920 S40network       {S40network} /bin/sh /etc/init.d/S40network start
1104 2944 sshd             /usr/sbin/sshd -f /etc/sshd_config
1000 1904 S60date          {S60date} /bin/sh /etc/init.d/S60date start
 604  948 sleep            sleep 900
 940 1372 proftpd          proftpd: (accepting connections)
4308 9468 smbd             smbd -D
1784 4528 nmbd             nmbd -D
 14m  16m python2.7        /usr/bin/python2.7 /programs/motioneye/motioneye.py
2300 9524 smbd             smbd -D
   0    0 cifsd            [cifsd]
 14m  61m motion           /usr/bin/motion -c /data/etc/motion.conf -n -d 1
1104 1888 S95motioneye     {S95motioneye} /bin/sh /etc/init.d/S95motioneye star
 568  964 getty            /sbin/getty -L tty1 115200 vt100
2448 5528 sshd             sshd: root@notty
 940 2324 sftp-server      /usr/libexec/sftp-server
2448 5532 sshd             sshd: root@pts/0
 528  948 sleep            sleep 5
1644 1952 sh               -sh
 592  948 sleep            sleep 5
 588  948 sleep            sleep 5
 624  952 ps               ps -o rss,vsz,comm,args

ccrisan, 2015-05-20:
Was the command run at a moment while the problem was happening? I mean it's clear that motion eats up most of the memory, but it's not that much.

borlandoflorida, 2015-05-20:
Yes, I ran the command when the cameras were showing as offline (grey boxes with text along the lines of "not seen since..."). It is baffling me because they were working up until the other day then suddenly stopped.

I'm actually going to try a fresh install on a different RPi as I want to use the B+ for something else because of the additional USB ports, so will be switching to an older Model B. I'll do a fresh install and see whether the issue is there, but if you have any suggestions then I'm all ears.

borlandoflorida, 2015-05-21:
Ok, I tried a fresh install and the problem is still there. I also couldn't get my Model B to boot at all, but I didn't have loads of time to try again. I can add both my cameras using the MotionPie interface, so they are being detected, but there is no live feed - just a greyed out box (not the one with the camera symbol) that sometimes flickers green and, even more rare, a few seconds of live feed will show on one of the cameras.

I suspect it could be a power issue, but as I do not have a powered USB hub I cannot verify. However the odd thing is that both cameras were working fine for several days and then all of a sudden they aren't. 

I've tried overclocking and altering the amount of memory allocated to the GPU, but nothing has an effect.

Bit odd, but surely if it was a power issue then the cameras would not have worked together in the first place, rather than fail a few days later?

guadacorte, 2015-05-21:
Borlandoflorida I purchased a 5.25 v PSU from modmypie thinking that it was power issue. Hasn't fixed the problem though.

borlandoflorida, 2015-05-21:
Do you mean an actual PSU or a powered USB hub for the cameras?

guadacorte, 2015-05-21:
They actual PSU. Have tried a powered hub as a test but it still failed. Not that its needed as there are enough onboard USB ports. At one time I thought it was the WiFi dongle but it still fails when cabled. I'll turn on logging and have a look.

borlandoflorida, 2015-05-21:
It's not the number of USB ports that could be the issue, it is the amount of power being drawn from the ports by the USB connected devices that could be. That's what I was getting at - i.e. if I connected my cameras to a powered USB hub and then connected the one USB cable to the Pi, the power demand should be lower, which would then tell me if it is a power issue or not.

I tried a wifi dongle and one USB camera, which worked. I tried a second USB camera, still with the wifi dongle, and then it failed. I then tried ethernet instead of the wifi dongle (thereby using two USB ports drawing power instead of three) and it worked. Until a few days ago when the live feeds turned grey.

borlandoflorida, 2015-05-21:
Ok, this is embarrassing...

I remembered that I switched the power supply for my RPi as part of a tidy up of cables, so I thought it might be that. Tried the previous power supply and hey-presto, everything is working again! Doh!

guadacorte, 2015-05-21:
Glad you got it sorted.

ccrisan, 2015-05-22:
So is the problem still there? @guadacorte, @dpatonsr I never got your output to `ps -o rss,vsz,comm,args` (while facing the Cannot allocate memory error).

borlandoflorida, 2015-05-22:
Sorry about the diversion! At least we have eliminated the power issue though...

ccrisan, 2015-05-22:
@borlandoflorida no worries. Unfortunately more users have reported similar issues and most of them have this Cannot allocate memory error in their logs. My assumption is that motion leaks memory in certain configuration, but what I've seen so far (regarding the `ps` command) the results are inconclusive. I need more cases and I need to find common configuration between the affected systems, to be able to identify when the (presumable) memory leak occurs.

dpatonsr, 2015-05-22:
I think I can recreate the situation where I was having the memory leak.
One of several things I did resolved the leak.  The most likely thing was
not adding a camera to another camera.  I saw the vsiz at over 75% for one
of the motion processes.  I had 2 cameras added to another camera when this
occurred.  I also had motion notifications on, etc. etc.  So I don't know
which feature was actually causing the issue.  I think I have a config file
saved from when I was set up in the way described above.

I was able to reboot the camera by sshing to it and rebooting.  The reboot
from the web page was not rebooting...
Thanks
Dave

guadacorte, 2015-05-22:
Ok. I have now setup a test system. I have enable debugging. Silly question but where is the log file store? thanks

ccrisan, 2015-05-22:
@dpatonsr having two cameras with motion detection enabled on 512MB could indeed push the memory usage to its limit. I never used more than one camera with a 512MB PI in "production".

dpatonsr, 2015-05-22:
All of my cameras are pi 2's.

guadacorte, 2015-05-23:
This seems prevalent in the log file after my second camera went off line
[2] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video1: No such file or directory

ccrisan, 2015-05-23:
@guadacorte well that clearly says the system somehow "lost" your second camera. Assuming that the second camera is a USB camera, the device somehow gets disconnected. I also assume that you're running the stable 20150331 version. Please install the latest prerelease (20150516) and try to see if the problem still occurs. If it does, please send me the dmesg.log and messages.log files.

rafi1975, 2015-06-30:
hi,
I just bought a brand new rpi2 with pi camera noir. Last night motioneye was installed and all working ok for ca.4-5 hours then these messages in logs and no camera output:

ERROR: failed to relay event: [Errno socket error] [Errno 111] Connection refused

Thanks,

Rafi

dpatonsr, 2015-06-30:
Hey Rafi
It is all in how you set up the configs. I have 3 Cameras.  They work fine
now.  Some things that the settings allow you to do are not reliable.  Best
to stay with a simple setup and slowly add capability that you are
interested in to see if it will work.

It is good it only takes a few hours to lock up.  You will be able to
figure it out faster.

I have only one camera added to each url.  Adding multiple cameras was
causing the lock up for me.

Turn off the motion detection to see if you get the lock up (for example)

rafi1975, 2015-06-30:
hi dpatonsr,

Many thanks for prompt reply.
I just switched off motion detection and see what will happen.
This is basic setup, only one camera.
I had similar setup on rpi1 B earlier with USB camera connected (supported
one) and it was crashing every couple of hours.
Hence I decided to go with the new rpi2.
I'll keep you posted.

Rafi

ccrisan, 2015-06-30:
A simple setup should work w/o problems. Please send me the log files so I can identify the possible issue.

rafi1975, 2015-06-30:
Hi,

What logs should I send?
I'm still noob :-)

Thanks,

Rafi

ccrisan, 2015-06-30:
See https://github.com/ccrisan/motionpie/wiki/Reporting-An-Issue.

dpatonsr, 2015-06-30:
I started with a B+ but as I observed and ccrisan confirmed, it is a bit
underpowered and leads to some frustration.  Since moving to the pi2 and
not using certain features I have been up 24x7 for several weeks.  ccrisan
also recommended not using the highest resolution due to cpu utilization.
I started midrange and bumped it up a bit without an issue.

rafi1975, 2015-06-30:
This is the last entry from motion.log

1] Selected palette YU12
[1] Test palette YU12 (1280x960)
[1] Error setting pixel format VIDIOC_S_FMT: Device or resource busy
[1] VIDIOC_TRY_FMT failed for format YU12: Device or resource busy
[1] Unable to find a compatible palette format.
[1] ioctl (VIDIOCGCAP): Inappropriate ioctl for device
[1] Could not fetch initial image from camera
[1] Motion continues using width and height from config file(s)
[1] Resizing pre_capture buffer to 1 items
[1] bind(): Address already in use
[1] Problem enabling stream server in port 8081: Address already in use
[1] Thread exiting

What logs should I provide if it crashes again, please?

Rafi

dpatonsr, 2015-06-30:
ccrisan will have to help you with the debugging.  I am a user.  looks like
you have been using a large format (1280x960).  I think there are 5 log
files created now.  save those for him.

I've noticed that the shutdown command on the UI doesn't always shutdown
the pi.  So I ssh over to it and do the shutdown that way.  You can also
check some of the pi stats while ssh'ed.  just doing a "top" has shown me
that a certain process is consuming all of the resources...

rafi1975, 2015-06-30:
Another crash:

un 30 15:43:30 raspberrypi motion: [1] Closing video device /dev/video0
Jun 30 15:43:34 raspberrypi motion: [0] motion restarted
Jun 30 15:43:39 raspberrypi motion: [0] Processing thread 0 - config file
/home/pi/motioneye/conf/motion.conf
Jun 30 15:43:39 raspberrypi motion: [0] Processing config file thread-1.conf
Jun 30 15:43:39 raspberrypi motion: [0] Motion 3.2.12 Started
Jun 30 15:43:39 raspberrypi motion: [0] Thread 1 is from thread-1.conf
Jun 30 15:43:39 raspberrypi motion: [1] cap.driver: "bm2835 mmal"
Jun 30 15:43:39 raspberrypi motion: [1] cap.card: "mmal service 16.1"
Jun 30 15:43:39 raspberrypi motion: [1] cap.bus_info:
"platform:bcm2835-v4l2"
Jun 30 15:43:39 raspberrypi motion: [1] cap.capabilities=0x85200005
Jun 30 15:43:39 raspberrypi motion: [1] - VIDEO_CAPTURE
Jun 30 15:43:39 raspberrypi motion: [1] - VIDEO_OVERLAY
Jun 30 15:43:39 raspberrypi motion: [1] - READWRITE
Jun 30 15:43:39 raspberrypi motion: [1] - STREAMING
Jun 30 15:43:39 raspberrypi motion: [1] Test palette YU12 (1280x960)
Jun 30 15:43:39 raspberrypi motion: [1] Using palette YU12 (1280x960)
bytesperlines 1920 sizeimage 1843200 colorspace 00000001
Jun 30 15:43:39 raspberrypi motion: [1] found control 0x00980900,
"Brightness", range 0,100
Jun 30 15:43:39 raspberrypi motion: [1] #011"Brightness", default 50,
current 50
Jun 30 15:43:39 raspberrypi motion: [1] found control 0x00980901,
"Contrast", range -100,100
Jun 30 15:43:39 raspberrypi motion: [1] #011"Contrast", default 0, current
31
Jun 30 15:43:39 raspberrypi motion: [1] found control 0x00980902,
"Saturation", range -100,100
Jun 30 15:43:39 raspberrypi motion: [1] #011"Saturation", default 0,
current 19
Jun 30 15:43:39 raspberrypi motion: [1] found control 0x0098090e, "Red
Balance", range 1,7999
Jun 30 15:43:39 raspberrypi motion: [1] #011"Red Balance", default 1000,
current 1000
Jun 30 15:43:39 raspberrypi motion: [1] found control 0x0098090f, "Blue
Balance", range 1,7999
Jun 30 15:43:39 raspberrypi motion: [1] #011"Blue Balance", default 1000,
current 1000
Jun 30 15:43:39 raspberrypi motion: [1] Resizing pre_capture buffer to 1
items
Jun 30 15:43:40 raspberrypi motion: [1] Resizing pre_capture buffer to 3
items

Motion process is running but no motioneye.

Rafi

ccrisan, 2015-06-30:
Please attach all the logs as indicated in the wiki article. Just portions of one log won't help me find the problem.

rafi1975, 2015-06-30:
Hi Calin,

I've been searching for the option to download logs in motioneye and can't
find.
I was able to locate motion.log in ~/motioneye/log
When I go to /etc/motion, there are no logs at all, just conf file.
Below is the link to messages and motion.log files.
I decreased resolution of the camera and motioneye keeps crashing. This is
very frustrating as the install is fresh - only motioneye and the required
components have been installed. Rpi was updated also.

http://wikisend.com/download/157302/messages
http://wikisend.com/download/185858/motion.log

Thanks,

Rafi

ccrisan, 2015-07-01:
It looks like you're not using motionPie. This place is only for motionPie. If you're using Raspbian + motionEye then the driver and hardware things are not under my control :)

rafi1975, 2015-07-01:
Hi,

I'm glad you've found what the issue is.
I'll get MotionPie loaded on an SD card and start testing with it.

Thanks a mill,

Rafal