ccrisan/motionpie #194: Cameras off-line after 20150331 update

Jonovich, 2015-05-04:
Found both my Camera image feeds were off-line (Motionpie was running though), and any changes to the settings would fail. Reboots from the Motionpie didn't fix this). However, I was suspicious of the update 20150331 that I applied the other day and in particular the: -

mount -o remount,rw /

from the changes that had to be applied to get the update to recognise.... So in for a penny and all that, I jumped on and issued: -

mount -o remount,rw /
mount -o remount,ro /
reboot

Low and behold, both Motionpies and their cameras were back on-line. ;)

Jono

ccrisan, 2015-05-05:
Please send me the motioneye.log file so I can see what actually happened back when it didn't work. The commands that you ran as a workaround don't make too much sense to me and I would really like to identify the root cause of your problem :)

ccrisan, 2015-05-05:
Mind you that you can (hopefully) download the motioneye.log file from the Expert Settings UI section.

ccrisan, 2015-05-05:
In your logs I was able to see "OSError: [Errno 12] Cannot allocate memory" many times. What PI models do you use?

Jonovich, 2015-05-05:
Both Model B’s, early ones, so 128Mb.

Cheers,

Jon.

Jonovich, 2015-05-05:
Should I drop the graphics memory down to free up more than 128Mb?

Cheers,

Jon

ccrisan, 2015-05-05:
On a128MB device, things can get pretty nasty and chances are that you run out of memory with motionPie. It's possible that python 2.7.9 (shipping with the latest version) is more memory hungry than its predecessor.

If you don't use a PI camera, you can reduce the GPU memory down to 16 MB and see if things get better.

closed 2015-05-05 by ccrisan