ccrisan/motionpie #195: Capture area

JankyLV, 2015-05-07:
Hello! I want to look for motion in a specific area of the picture. I suppose it could be done through "Extra Motion Options", but I have no idea how. I tried "area_detect", but it didn't work. Any help would be much appreciated.

ccrisan, 2015-05-07:
I don't think you can "look" into a specific area of the picture. I believe you can only tell the motion detection algorithm to analyze a specific area. This is not currently possible using the UI but you could, as you have identified yourself, manually specify additional motion arguments in the Extra Motion Options.

I must confess I never toyed with these detection area options, so your only option is to read the [docs](http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigFileOptions).

JankyLV, 2015-05-07:
I just don't know what syntax i'm supposed to use in that field.
Or maybe my best bet is to edit the motion.conf itself?

The reason why I'm trying to do this, is that I want to capture even the slightest movement, but trees in the wind are kind of ruining everything :-1: 
For now I have the camera pointing at the ground, and the driveway is just at the corner of the frame, but still - the motion capture is triggered by the trees.

ccrisan, 2015-05-07:
I don't know the syntax. You can directly edit your /data/etc/thread-1.conf file, or you can use Extra Motion Options. It should have the exact same effect.

ccrisan, 2015-05-16:
Duplicate of https://github.com/ccrisan/motionpie/issues/170

closed 2015-05-16 by ccrisan