ccrisan/motionpie #196: Recording to SMB (NAS)

Jasimo, 2015-05-08:
Hi, as Iam new to motionpi I have a question regarding SMB recording.
When I set the recording to a NAS via SMB (NFS is not possible, isn´t it?) what happens wenn the NAS failes or the network connection is interrupted. Is there any buffering of the recorded files locally on the PI?
rgs
Jan

ccrisan, 2015-05-08:
When the NAS fails, your video stream is interrupted and everything is lost. There's no fallback and there's no easy way to implement one.

motionPie does not support NFS.

Jasimo, 2015-05-08:
would be a nice feature. But it should be possible to record locally and copy all records older then x to the nas to keep the local storage free?

closed 2015-05-16 by ccrisan