ccrisan/motionpie #197: Bright Pi

n3dv3d, 2015-05-12:
Hi,can i use Bright Pi LEd with motionpie?how can i install i2c-tools?

Thank you

ccrisan, 2015-05-12:
You cannot install anything on motionPie.

shawaj, 2015-06-22:
Do you want me to send you one, so that it can be integrated into the software?

ccrisan, 2015-06-23:
If there are a few simple and small packages that need to be installed, I would be happy to include them. There's no need to send me such a camera - I believe we can manage to make it working by remote testing.

shawaj, 2015-06-24:
It is not a special camera, it is a bright white and infrared light module
that allows you to illuminate the scene around the camera.

It requires i2c to be turned on on the Pi, and then it needs to issue some
commands to turn the light on/off

Is it worth having some kind of button / time based mechanism to turn the
lights on?

Thanks

ccrisan, 2015-06-25:
I'll make sure to include i2c support. What packages need to be installed? I won't integrate it with the UI but anyone wishing to play with it can use the /data/etc/userinit.sh and crontab to do whatever they wish.

shawaj, 2015-06-25:
I'll collate all the details for you.

Would it be easiest if our API / command line tools were all in a deb
package or something?

ccrisan, 2015-06-25:
I'm not sure what you mean by "our API". Buildroot works best with simple downloadable archives. If you can put together a list of files (or an acrhive) to be downloaded and "installed" into some folder on the root filesystem, I could add that to motionPie for the RPi.

rauryn, 2015-07-19:
The details of the i2c package are in this link:

http://pkgs.org/download/i2c-tools

If if could be added, that would be great. It would let me use a UPS with the Pi for added security.

ccrisan, 2015-07-19:
i2c support just landed in the next version that will hopefully be released later today.

closed 2015-07-19 by ccrisan

Foeke, 2015-10-11:
For the use of Bright pi, more has to be done then just the availability of i2c-tools
add in /etc/modules two lines
    i2c-bcm2708 
    i2c-dev

 add in /boot/config.txt two lines
   dtparam=i2c1=on
   dtparam=i2c_arm=on

I cannot add these lines because everything is read only.
Maybe you could add these in a next release.

The use of the LED's is explained on this URL https://www.pi-supply.com/bright-pi-v1-0-code-examples/
Maybe it is possible to trigger the LED's (white and/or IR) from the Motionpie GUI.

I have 3 camera's operating with your beautiful software.

ccrisan, 2015-10-11:
https://github.com/ccrisan/motionpie/wiki/FAQ#why-do-i-get-read-only-file-system-when-trying-to-edit-a-file

raspberrycoulis, 2016-01-24:
Has anything progressed on this? I'm trying to use the Bright Pi addon as well, but I cannot seem to get i2c working? When I run i2cdetect -y 1 (or 0) via SSH, I get "Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory".

I've added i2c-bcm2708 and i2c-dev to a newly created modules file (in /data/etc/ as recommended here (https://github.com/ccrisan/motioneyeos/wiki/Tweaks), but no joy.

My userinit.sh script looks like this:

i2cset -y 1 0x70 0x00 0x5a &
i2cset -y 1 0x70 0x02 0x32 &
i2cset -y 1 0x70 0x04 0x32 &
i2cset -y 1 0x70 0x05 0x32 &
i2cset -y 1 0x70 0x07 0x32 &

Which is what is required to turn the LED's on the Bright Pi on / off etc. When I check the boot.log, it says it can run the userinit.sh with success, but when I try and run any of the i2c related inputs at the command line via SSH, I get the error relating to /dev/i2c

Any suggestions?

raspberrycoulis, 2016-01-24:
I think I may have answered my own question.

I was looking through the file system and found the i2c reference in /sys/bus...

I then added the following to the modules in /data/etc

bcm2708_i2c.1

And it works!