ccrisan/motionpie #198: Spaces in passwords make it unable to SSH into motionPie

BeanBagKing, 2015-05-13:
To recreate, log in and set an admin password via the UI to anything containing spaces, for example, "pass word". Reboot and try to SSH to the box. You should get the following error.

root@box:~# ssh 10.0.0.17
Welcome to mp-xxxxxxx!
root@10.0.0.17's password: 
Permission denied, please try again.
root@10.0.0.17's password: 
Permission denied, please try again.
root@10.0.0.17's password: 
Permission denied (publickey,password,keyboard-interactive).
root@box:~#

I know this isn't a problem on other linux boxes. It's probably low hanging fruit for motionPie, but I hope it gets solved one day because I use passphrases for a lot of things.

Thanks.

ccrisan, 2015-05-13:
I will take a look into this.

ccrisan, 2015-05-31:
The problem has been fixed. The fix will be included in the next version.

closed 2015-05-31 by ccrisan