ccrisan/motionpie #199: No Picture using Pi Noir with Motionpie

Chip-DFIR, 2015-05-15:
Hi 

I have setup and installed using the motionpie-raspberrypi2-20150331.img.gz image

I have the webgui up and running and it has detected the camera but the screen shows:

[image]

and on the monitor I have plugged into the Pi  I get an error that refreshes every 30 seconds or so with different numbers at the start but ending:

bcm2835-v412: error 0 waiting for frame completion

Each time the error repeats the red led on the noir blinks off then on.

Has anyone else had this issue or know of a fix please?

ccrisan, 2015-05-15:
Does it work with raspbian?

Chip-DFIR, 2015-05-15:
I'm not sure I have only tried your raspberry pi2 image as I was following the guide.

Should I install Raspbian then install from source?

ccrisan, 2015-05-15:
Nope. Just install raspbian and see if other v4l2 applications work with the camera, so we can rule out a hardware fault.

Chip-DFIR, 2015-05-15:
What other applications should I try? Sorry quite new to the Pi and the camera

ccrisan, 2015-05-15:
First try to see if [raspistill](https://www.raspberrypi.org/documentation/usage/camera/raspicam/raspistill.md) works.

Then you can `sudo modprobe bcm2835-v4l2` to have the v4l2 driver loaded and you can `sudo apt-get install camorama` to test the v4l2 interface.

Chip-DFIR, 2015-05-15:
When I try Raspistill I am getting the following error:

mmal: No data received from sensor. Check all connections, including the sunny one on the camera board.

I have reseated the cable with no joy though.

ccrisan, 2015-05-16:
Then I suspect a hardware problem.

closed 2015-05-16 by ccrisan