ccrisan/motionpie #200: Having problem adding ip camera.

db414813, 2015-05-17:
Hello,

So I am trying to add a ip camera to MoPie. The ip cam is an old android phone running ip camera app. I have successfully added to zoneminder before. Now, I add the network camera in mopie (thats a good nickname for the os) I specify the url as http://192.168.1.4:8080/video which the ip cam app specifies is the mjpeg stream of the app. In the camera section of the add camera menu it recognizes it as a "MJPEG Network Camera". I select ok. Then the cam view appears gray with "UNABLE TO OPEN VIDEO DEVICE". Not sure how to add it. There is no password or username made for the ip android camera, so I did not enter one. I am not sure what to do here. Any ideas?

db414813, 2015-05-17:
I tried adding a password and username, but it just yielded the same gray window.

ccrisan, 2015-05-17:
Hi,

First of all, I need to see the logs. More precisely, I'm interested in motion.log. You can download it from the UI (Expert Settings section) and paste it using [pastebin](http://pastebin.com/).

Secondly, I would like to know the exact android app (possibly with settings) that you're using to turn it into an IP cam; I'd like to reproduce the problem myself.

db414813, 2015-05-17:
Well I got rid of mopie. I'm on raspian to use the web can interface now,
so no logs. The app I used was IP webcam. I'm sure you can easily replicate
the results. Let me no if it works for you please?

ccrisan, 2015-05-18:
MotionEye always saves a motion.log file. That's the one I'd like to see.

ccrisan, 2015-05-18:
What version of motionPie are you using?

db414813, 2015-05-18:
I overwrite the sd card with wheezy.

db414813, 2015-05-18:
I originally ran motionpie not motioneye. When I get some time I will
install motion eye and replicate my results. I am in the middle of moving.

db414813, 2015-05-18:
Oh, and I used your latest release.

dkelley810, 2015-05-23:
i ran into this issue and had to use the Surveillance Username and password to get it to work.

ccrisan, 2015-05-23:
@dkelley810 the issue here is a bit different than yours. @db414813 is trying to add an IP camera to motionPie. You're trying to use motionPie as an IP camera. Correct me if I'm wrong.