ccrisan/motionpie #201: 3-5sec lag, high cpu load

tuxedo0801, 2015-05-23:
Hi there,
before I came to motioneye, I was using my Raspberry Pi B with CSI camera + a modified version of mjpeg-streamer (which enabled me to use the cam directly without raspistill/raspivid, as there was no v4l support).
The performance was quite okay: 1280x720 resolution with about 10..15fps with a CPU load somewhere below 60%. Almost no lag, maybe <300ms or so..
Now I updated to latest stable motionpi version, which I assume, uses the new v4l support for CSi cam.
The performance is really bad:
1280x720 resolution with configured 10fps, video streaming disabled, ligth switch detection disabled, automatic brightness disabled, the rest is at default.
I now have 100% CPU load (~60% for motion and ~40% for motioneye) and a lag somewhere around 3-5sec ...
What's going wrong? Do you face the same performance on your side? How can I speed things up? I already tried lowering the resolution, wich results in lower CPU load. But lag is still very bad. And honestly: I don't want to use old-fashioned 800x480 (or even lower) resolution if there is a 5MP cam available ... 1280x720 is a "must have".
br, Alex

tuxedo0801, 2015-05-23:
I pushed my old mjpeg-streamer to the motionpie pi and started it...

mjpeg-streamer eats ~40% cpu with 1280x720 @ about 20fps. Again, almost no lag if stream is watched directly.

As soon as motioneye has a camera  configured, motion eats CPU. No matter if I configure 2fps or disable "motion detection" at all.

ccrisan, 2015-05-23:
Motion is known not to be the best choice when it comes to simple mjpeg streaming (in terms of performance).

The good news is that I'm working on an addon for motionPie on RPi that will allow you to transform your motionPie into a fast and performant mjpg streamer, with the click of a button. The UI will allow you to configure a few parameters and it will use a [streamEye](https://github.com/ccrisan/streameye) as a backend. The bottleneck will hopefully be the network connection, not the CPU.

Until then, your only option is to stay with mjpeg streamer.

tuxedo0801, 2015-05-23:
I'm not quite sure, but: If I don't use the mjpeg-stream of motion (disabled in motioneye) at all: The CPU load is still very high. Example:

1280x720 @ 2fps, streaming completely disabled.

cpuload:
motion: ~60% avg.
motioneye: ~5% avg.

Question is: What is motion actually doing? There's no motion detectione enabled. Really? No, motion is streaming to localhost. I guess it's used for motioneye to get a picture? 

mjpeg-streamer is quite more efficient for streaming. Why not use mjpeg-streamer to provide the stream and use motion (without streaming) to detect "motion"?

Are you sure that streamEye provides better performance as mjpeg-streamer (whoch obviously does exactly the same thing?)? Are you really using the GPU to get&provide the videostream?

What about "motion detection with help of gpu"?

Anyhow: How far is your development for this solution?

ccrisan, 2015-05-24:
I believe motion eats up a lot of CPU when it captures images and encodes/decodes JPEG. So there's not much to do about it to reduce the CPU usage with motion, except for reducing the frame rate or resolution.

I needed a way to capture jpegs taken and compressed directly by the PI's GPU, in video mode (as opposed to the stills mode, which is slower). A series of such jpegs is the basis of a MJPEG stream. This is why I wrote a small script called [raspimjpeg.py](https://github.com/ccrisan/streameye/blob/master/extras/raspimjpeg.py) that behaves similarly to raspistill but uses the video mode.

Now given this stream, I needed to feed it to an HTTP server of some sort, so that the MJPEG can be accessed through HTTP. This is why I developed [streamEye](https://github.com/ccrisan/streameye).

I could have used mjpeg-streamer but I preferred my own way so that many of the parameters could be tweaked using the UI.

Yes, I'm sure the video stream will be provided by the GPU.

Motion detection with the help of the GPU is not my field and I don't wish to start developing such a feature on my own. Especially given that the entire motionPie/motionEye solution is based on motion, which does all the hard work here.

The development of my proposed solution is close to its release. I need a weekend or two to code the UI interaction and run some tests. I'm not promising anything though.

I hope I answered all your questions.

tuxedo0801, 2015-05-25:
Hi,
 > So there's not much to do about it to reduce the CPU usage with motion, except for reducing the frame rate or resolution.

Hmm, okay. Maybe motion is not that bad. maybe the Raspi has just not enough power ... I'm thinking about - for testing purpose - running mjpeg-streamer in the pi and runnint motion-eye on my debian machine ... But actually it would be better to just have one machine running for security webcam.

> I could have used mjpeg-streamer but I preferred my own way so that many of the parameters could be tweaked using the UI.

That makes sense, of course.. *thumbsup*

> Motion detection with the help of the GPU is not my field and I don't wish to start developing such a feature on my own. 

Yes, doing it from scratch is very hard.

> The development of my proposed solution is close to its release. I need a weekend or two to code the UI interaction and run some tests. I'm not promising anything though.

If you have something to (beta-)test, just let me know. 

br,
Alex

ccrisan, 2015-06-23:
@tuxedo0801 the solution is ready (to be tested).

closed 2015-06-23 by ccrisan