ccrisan/motionpie #202: More camera woes

jepolch, 2015-05-23:
I have a persistent problem with every version of motionPie when adding cameras. I bought a bunch of small "eyeball" cameras, mostly Logitech. I also have a Raspberry Pi cam on my Pi 2 B+. The scenario is always the same. When I start motionPie for the first time on a new install, the three cameras (two Logitech, one RPi) can be added, but come up with low resolution settings by default, like 320x240. I want them to all to be at least 640x480. I change the first two to 640x480 and everything is going well. As soon as I change the third camera to 640x480, the camera initially shows the new setting, but then motionPie seems to lock up. The clocks in all the views stop running. If I reboot at this point motionPie will not start. It doesn't matter which camera I switch to 640x480 in the third instance, motionPie will lock up. 

Another issue: motionPie was running fine with two cameras at 640x480 and one at 320x240. I added a fourth USB camera to the setup. I opened the UI and as soon as I added the fourth camera motionPie locked up. It would not restart even after I removed the fourth camera. I had to image a new SD card. Maybe it was a coincidence, but the SD card that that was in the Pi at the time this incident occurred could no longer be recognized by my PC. Again, maybe it's a coincidence, but this is not the first time motionPie crashed and seemed to ruin the SD card.

Is anyone else having similar issues?

I really like motionPie and will stick with it as it's the best solution for a low-cost security monitoring system, but these camera issues are very annoying.

ccrisan, 2015-05-24:
1) You're on a PI 2 B or a PI B+? There's no PI 2 B+.
2) Which is the third camera? A logitech or the RPi Camera? Does it matter? Have you tried a different order?
3) Are you sure your power supply is powerful enough to handle everything? What current rating does it have? Have you tried with a different power supply?
4) Is your system overclocked? Have you tried downclocking?

Normally the card doesn't get corrupted unless there are some power-related issues.

jepolch, 2015-05-24:
1) Sorry, it's a Pi 2 B.
2) The third camera is random - the order changes each boot. So no, it doesn't seem to matter.
3) The cameras are plugged into a powered USB hub and that plugs into the Pi. The Pi has a 2.1 amp power supply. When I have a little more time I'll try plugging one of the Logitech cameras directly into the Pi and try the resolution settings again.
4) UI shows it's overclocked to medium - 900 - but isn't that the normal frequency for the Pi 2? I did some experimenting. When I overclocked to high, neither of the Logitech cameras would display a picture in the UI. No clock setting allows two Logitech cameras to be set to 640x480 at the same time.

Thanks for your time and have a good day.

ccrisan, 2015-05-24:
3) Please do, as it seems to me that it's somehow related to power.
4) Yes, 900 is the default for a PI2. I was curious if a lower frequency (700) would prevent the problem from triggering.

Could you please send me the logs after such an "incident"? See [Reporting An Issue](https://github.com/ccrisan/motionpie/wiki/Reporting-An-Issue#attach-log-files) if you don't know how to obtain the log files.

jepolch, 2015-05-24:
I set camera 2 (Logitech) to 640x480 and the cameras all froze. At this
point all three cameras (two Logitech, one Pi) are set to 640x480.

http://wikisend.com/download/833774/jepolch logs.zip

jepolch, 2015-05-24:
Here's the motioneye.log after I set camera 2 to 320x240 and all the
cameras "unfroze".

http://wikisend.com/download/690856/motioneye(1).log

ccrisan, 2015-05-24:
It looks like the motion daemon hangs after a while (or when you set 640x480). Could you please turn on the Enable Debugging option in Expert Settings, repeat the procedure to reproduce the issue and send me the logs again?

ccrisan, 2015-05-24:
Also, could you try the prereleased [20150516](https://github.com/ccrisan/motionpie/releases/tag/20150516) version and see if the problem persists? If yes, I'd prefer the logs from this version.

jepolch, 2015-05-24:
OK, I'll do those things, but it won't be today. I have used the prereleased version before this current installation. It was the one that locked up and that's when the SD card was ruined. When I reinstalled I went back to the current release. I'll make a new SD with the prerelease as soon as I can.

ccrisan, 2015-05-24:
The prerelease version has dmesg.log and boot.log which will help identifying USB-related issues.

jepolch, 2015-05-24:
I'll send the logs after I install the prerelease.

ccrisan, 2015-05-24:
The new stable version [20150524](https://github.com/ccrisan/motionpie/releases/tag/20150524) is already out, by the way.

jepolch, 2015-05-24:
Thanks. I'll install that version.

jepolch, 2015-05-25:
I installed 201524 and changed all cameras to 640x480. First, the Pi cam, then one Logitech, then the second Logitech. All cameras froze and motionPie became very sluggish. Both Logitiech cameras stopped displaying an image in the UI, but the Pi camera still showed, but the clock was frozen.

jepolch, 2015-05-25:
I installed 201524 and changed all cameras to 640x480. First, the Pi cam,
then one Logitech, then the second Logitech. All cameras froze and
motionPie became very sluggish. Both Logitiech cameras stopped displaying
an image in the UI, but the Pi camera still showed, but the clock was
frozen.

http://wikisend.com/download/841516/jepolch logs 2.zip

ccrisan, 2015-05-25:
It might just be too much for the PI. I don't have 3 cameras to test this scenario. Nevertheless, please send me all the log files, including boot.log and dmesg.log.

jepolch, 2015-05-25:
That's what I'm afraid of. I sent you the logs. By the way, which camera(s) do you use for testing?

ccrisan, 2015-05-25:
Where are the logs? How did you send them to me? I use a simple RPi camera and some cheap and stupid USB camera.

jepolch, 2015-05-25:
Two or three emails above. I'm not at home so I can't resend the link now.

ccrisan, 2015-05-25:
I have some logs from your setup but they don't have boot.log and dmesg.log. I need the logs generated by 20150516 or 20150524.

jepolch, 2015-05-25:
Sorry, I tried to send it to you in email. http://wikisend.com/download/841516/jepolch logs 2.zip

ccrisan, 2015-05-25:
Did your issue occur during the period from the last boot? The dmesg.log file only contains details from the current boot, as opposed to others that are appended to each boot.

jepolch, 2015-05-25:
Here's one from the current boot with the cameras frozen. I'm not sure if you wanted me to reboot and send the log, or send it from the current boot with the cameras frozen. http://wikisend.com/download/709036/jepolch_dmesg.log

ccrisan, 2015-05-26:
Hi,

I took some time to analyze your problem. There's one line in your dmesg.log that raises questions: `uvcvideo: Failed to resubmit video URB (-1)`. After some googling it appears that there are many Logitech UVC devices that exhibit this behaviour. For example [this](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/729627) bug was never fixed. [This](https://groups.google.com/forum/#!searchin/beagleboard/UVC/beagleboard/d5DzbjAI1oU/yTgyJIHl86cJ) problem has been reported with a Logitech camera on a BeagleBone board and [this one](https://www.mail-archive.com/linux-uvc-devel@lists.berlios.de/msg00718.html) says that it would be a hardware bug.

My point is that there's not much I can do about it as it seems to be a common problem among Logitech cameras on Linux; they appear to hang in certain conditions. My suggestion is to try other resolutions, try to see if the problem can be reproduced on Raspbian (with motion/motionEye or other V4l2 software) and of course look for driver parameters and fixes online.

jepolch, 2015-05-27:
Thanks very much for taking the time to look into it Calin. Yes, one of the Logitech cameras is identified as a UVC camera by motionPie. I was also hoping to hear from other users to see how they handled this type of issue when it occurs. Change cameras? Connect directly to Pi or USB hub? I've been busy for a couple of days, but I will  move one of the Logitech cameras from the USB hub and plug it directly into the Pi and update this with more info tomorrow. Thanks again.

jepolch, 2015-06-04:
Thanks for keeping this issue open. It's taken me longer than I thought, but I finally got around to plugging both Logitech cameras into the Pi. So all three cameras, including the Raspi cam are operating from the Pi. Still, only one Logitech camera can be set to 640x480 at one time, not both. The Pi is maxed out. Thanks and keep up the good work.

closed 2015-06-04 by jepolch