ccrisan/motionpie #204: Question about motion binaries

Jasimo, 2015-05-27:
Hi, what are the differents between the motion binaries:
motion
motion-mmal
motion-mrdave

ccrisan, 2015-05-27:
Hi,

* `motion` is the "official" [SVN](http://www.lavrsen.dk/foswiki/bin/view/Motion/MotionSubversion) version of  motion
* `motion-mmal` is the [fork](https://github.com/dozencrows/motion/tree/mmal-test) adapted for the `MMAL` raspberrypi camera interface - it won't work with V4L2/USB cameras
* `motion-mrdave` is another [fork](https://github.com/Mr-Dave/motion) that is supposed to be better than the official SVN version and that incorporates many bug fixes and improvements

To be honest I didn't play too much with the last two variants, I just included them in case users needed them. I'm waiting for feedback from the community especially on `motion-mrdave`; it can become the default one day.

oopeekoo, 2015-05-27:
Doesn't the motion-mrdave offer support for RSTP cameras? I'd like that! How do I activate this fork, replace '/usr/bin/motion' with '/usr/bin/motion-mrdave' in expert settings and reboot?

ccrisan, 2015-05-27:
Yes, motion-mrdave has support for RTSP cameras but motionEye doesn't. I don't know how those work and I don't own one so I couldn't add support in motionEye.

dattanchu, 2015-07-15:
I have a few different RTSP cameras. I would like to start using them in RTSP mode. When you say motionEye doesn't support RTSP, can you give some background on where I can start looking to add support for it?

Is motionEye only the web front end for configuring and launching these services?

ccrisan, 2015-07-15:
MotionEye doesn't support RTSP cameras in the sense that it cannot detect and add a RTSP camera. If you manually configure your network camera in motion.conf, it might just work. I don't have a RTSP camera to test, nor do I know how it actually works (apart from what I read on Wikipedia).

If you wish to test it, I suggest using the `/usr/bin/otion-mrdave` binary which supports these RTSP cameras, in theory. You could then add a normal network camera and then edit the /data/etc/thread-[x].conf by replacing the netcam_url setting to whatever URL your RTSP camera has.

Keep me posted.

dattanchu, 2015-07-26:
Changing the motion binary to motion-mrdave. Then change the correct thread conf's netcam_url to the rtsp URL that I have confirmed to work with VLC. I had to restart motionpie via reboot. Motionpie now get the rtsp feed. The aspect ratio is off. Instead of coming as wide-screen, it's defaulted to 4:3. Is there a configuration in the conf file that helps me set the aspect ratio?

dattanchu, 2015-07-26:
I looked at the recorded avi and the aspect ratio is also off. It must be a motion-related option and not what is being displayed in motionpie.

dattanchu, 2015-07-26:
I found the config page of motion. After setting width and height to 720P (what my stream setting is in my camera), the image comes out great.

http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigFileOptions

ccrisan, 2015-08-09:
I'm currently working on implementing support for RTSP netcams in motionEye/motionPie. The biggest problem is that, as I don't own a RTSP-enabled camera, I'm using VLC to simulate one.

Things are working fine with VLC, but I would like to test it with a real RTSP camera. The question for you guys is, would you be able to configure one of your RTSP cameras for remote access, so I can use it for testing?

dattanchu, 2015-08-10:
I have set up a dynamic dns with a visitor username and password for your testing. I will message you on Google chat.

ccrisan, 2015-08-10:
@dattanchu thank you for the RTSP link.

closed 2015-08-10 by ccrisan

jonj27, 2016-02-23:
Hi, 

Were you able to implement rtsp in MotionPie? I am looking to use it and would like to know how to configure to use my rtsp camera.

Thanks in advance

ccrisan, 2016-02-23:
[MotionEyeOS](https://github.com/ccrisan/motioneyeos) has some experimental RTSP support. Please switch to motionEyeOS.

jonj27, 2016-02-23:
Many thanks. I will do that.