ccrisan/motionpie #205: Clarification on how to incorporate multiple cameras

dhalp, 2015-05-27:
I just wanted to clarify on how I could include multiple input cameras.  Could I just use a bunch of raspberry pies with a camera module plugged in on the same network?  I did not see much documentation on how one should incorporate multiple camera modules from around a facility.

ccrisan, 2015-05-28:
I agree there's not much info on how to use multiple cameras. I am planning on writing one, but after I finish the work on the last piece of the puzzle - a fast RPi-based IP camera.

The way you should normally deal with multiple cameras is to install motionPie on as many SBCs as you wish (RPI, RPI2, BPI, whatever) and use a separate and more powerful Linux machine with [motionEye](https://bitbucket.org/ccrisan/motioneye/) to manage them together.

You could only enable motion detection & everything on the powerful (multicore) Linux machine, while using motionPie machines only as IP cameras. Disabling motion detection and advanced features on the IP cameras will improve the streaming performance (especially on RPI1).

dhalp, 2015-05-28:
Awesome! Thanks.

mixmediasec, 2015-07-28:
Can you please explain to a noob like me, what do you mean by a linux machine. and how do i install motionEye so i can use 2 rasp pi 2 that run motionpie. Please and thank you

ccrisan, 2015-07-28:
A linux machine is any computer capable of running a recent Linux distribution. See [this guide](https://bitbucket.org/ccrisan/motioneye/wiki/Installation%20Guide) to see how to install motionEye on Linux.

ccrisan, 2015-07-28:
I have put together a small wiki page to help people decide what usage scenario works best for them, given the relatively large possible configurations. See [Usage Scenarios] (https://github.com/ccrisan/motionpie/wiki/Usage-Scenarios).

closed 2015-07-28 by ccrisan

mixmediasec, 2015-07-29:
hello, i set up a rasp 2 to run motioneye, but when it typed in
./motioneye.py it kept telling me i needed to update python tornado to 3.1
or better. I then tried to manual install but i got the fooling errors

error: could not create '/usr/local/lib/python3.2/dist-packages/tornado':
Permission denied

Command /usr/bin/python3.2 -c "import
setuptools;__file__='/home/pi/motioneye/build/tornado/setup.py';exec(compile(open(__file__).read().replace('\r\n',
'\n'), __file__, 'exec'))" install --single-version-externally-managed
--record /tmp/pip-ywtpdn-record/install-record.txt failed with error code 1
in /home/pi/motioneye/build/tornado

can you please help. I am tryimg to learn on my own. But times like these i
reach out for help when i did everything on the instructions

ccrisan, 2015-07-29:
How did you try to manually install tornado? Did you use `sudo`?

mixmediasec, 2015-07-29:
Yes. Sudo aptitude install python-tornado

mixmediasec, 2015-08-10:
Was i wrong using sudo, can you please help?

ccrisan, 2015-08-11:
`sudo apt-get install motion ffmpeg v4l-utils python-dev python-pip libssl-dev libjpeg-dev libcurl4-openssl-dev`

`sudo pip install --upgrade tornado==4.1 jinja2 pillow pycurl`

mixmediasec, 2015-08-15:
i am now getting the following error:

ImportError:: No module named PIL

mixmediasec, 2015-08-16:
I have figured out how to install motioneye, I do not understand the
instructions on how to autostart, plz help.