ccrisan/motionpie #206: Invalid Credential error

kiuuby, 2015-05-27:
Hi,im getting a invalid credentials error when im trying to add a webcam. I can login the first time,change the password for admin and user, use ssh , but when i try to add a camera it gives this error. I've tried with my password, admin/root , the pi serial , leave it blank ,i tried everything but nothing. Can you help me?

oopeekoo, 2015-05-28:
Same here with Microsoft LifeCam HD. Switched to Logitech HD camera, no credentials asked anymore.

ccrisan, 2015-05-28:
Could you please send me the motioneye.log file and a screenshot of the Add Camera dialog when the name of the camera is visible? It might have to do with some special characters in the camera's name.

kiuuby, 2015-05-28:
I've got the same camera! right now im away and i cant upload the log or the screen sorry

oopeekoo, 2015-05-28:
Troublesome Microsoft camera showed up with full model name, working Logitech is just "V4L2 Camera".

ccrisan, 2015-05-28:
It might be the special (R) character in the camera name. I'll investigate the problem and come back with an answer.

wbrownp, 2015-05-31:
Hi, 
Just wanted to leave a quick note to say that I am too experiencing the same issue as oopeekoo. Trying to add a Microsoft Lifecam 3000 and credentials are not being accepted. Thanks

ccrisan, 2015-05-31:
[This](https://bitbucket.org/ccrisan/motioneye/commits/413e5ee41b6e688992cb9a77a10335a7c6eea1a1) motionEye commit fixes issues with Microsoft (R) cameras. The fix will be included in the next release.

closed 2015-05-31 by ccrisan