ccrisan/motionpie #208: ipcam freeze after change name

Master2Blaster, 2015-05-30:
hi there,

i am using a mpeg ipcam. 
When i add the cam at first time, it works ok.
But when i change one setting (camera name), it freezes (unable to open video source)
if i remove the cam and add it again it is working again.

ccrisan, 2015-05-31:
Please provide some [log files](https://github.com/ccrisan/motionpie/wiki/Reporting-An-Issue#attach-log-files) so I can identify the underlying problem.

dgmckelvie, 2015-08-09:
I have the same issue using dlink ip cameras.  initial setup of camera works, but then I changed the motion notification.
only diff was between the thread-2.conf files
BEFORE:
# @working_schedule_type outside
# @webcam_resolution 100
# @network_password
# @enabled on
# @name Camera2
# @network_server
# @id 2
# @motion_detection on
# @network_username
# @storage_device custom-path
# @preserve_movies 0
# @network_share_name
# @preserve_pictures 0
# @working_schedule
# @webcam_server_resize off

snapshot_interval 0
minimum_motion_frames 1
stream_motion off
target_dir /data/output
text_double on
netcam_userpass admin:mckelvie
stream_auth_method 0
stream_quality 85
threshold 2000
quality 85
text_left Camera2
post_capture 4
noise_level 32
framerate 2
netcam_url http://192.168.1.81/image/jpeg.cgi
pre_capture 2
noise_tune on
picture_filename
locate_motion_style redbox
locate_motion_mode off
stream_maxrate 5
output_pictures off
stream_localhost off
ffmpeg_variable_bitrate 0
ffmpeg_video_codec msmpeg4
movie_filename %Y-%m-%d/%H-%M-%S
auto_brightness off
text_changes off
on_event_end
stream_port 8082
rotate 0
text_right %Y-%m-%d\n%T
on_event_start
ffmpeg_output_movies off
ffmpeg_bps 44000
lightswitch 50
event_gap 10
max_movie_time 0
emulate_motion off
snapshot_filename
==============================================
AFTER:
# @working_schedule_type outside
# @webcam_resolution 100
# @network_password
# @enabled on
# @name Camera2
# @network_server
# @id 2
# @motion_detection on
# @network_username
# @storage_device custom-path
# @preserve_movies 0
# @network_share_name
# @preserve_pictures 0
# @working_schedule
# @webcam_server_resize off

snapshot_interval 0
minimum_motion_frames 1
stream_motion off
target_dir /data/output
text_double on
netcam_userpass admin:mckelvie
stream_auth_method 0
stream_quality 85
threshold 1999
quality 85
text_left Camera2
post_capture 4
noise_level 31
framerate 2
netcam_url http://192.168.1.81/image/jpeg.cgi
pre_capture 2
noise_tune on
picture_filename
locate_motion_style redbox
locate_motion_mode off
stream_maxrate 5
output_pictures off
stream_localhost off
ffmpeg_variable_bitrate 0
ffmpeg_video_codec msmpeg4
movie_filename %Y-%m-%d/%H-%M-%S
auto_brightness off
text_changes off
on_event_end /programs/motioneye/eventrelay.py stop %t
stream_port 8082
rotate 0
text_right %Y-%m-%d\n%T
on_event_start /programs/motioneye/eventrelay.py start %t; /programs/motioneye/sendmail.py "smtp.gmail.com" "465" "xxxxx@gmail.com" "XXXXXX" "False" "xxx@gmail.com" "motion_start" "%t" "%Y-%m-%dT%H:%M:%S" "2"
ffmpeg_output_movies off
ffmpeg_bps 43008
lightswitch 50
event_gap 10
max_movie_time 0
emulate_motion off
snapshot_filename

stream_authentication user:mckelvie
netcam_keepalive on
netcam_tolerant_check on

dgmckelvie, 2015-08-09:
motion.log is continuously logging:
[2] [ERR] [NET] netcam_read_first_header: Error sending 'connect' request:

ccrisan, 2015-08-09:
Thanks for your before/after config files. Thanks to these, I believe I found the root cause of the problem: the last two directives are added only *after first settings change*, not being present when the camera is added. They must cause some issues with your netcam. I'll dig deeper into this and find a proper solution.

dgmckelvie, 2015-08-09:
yes, I removed the last 3 lines from the conf that didn't work and the camera came back.

dgmckelvie, 2015-08-10:
FYI I tried just removing 'netcam_keepalive on' and it works

ccrisan, 2015-08-16:
The problem has been fixed upstream (in motionEye). The fix will be included in the next version.

closed 2015-08-16 by ccrisan