ccrisan/motionpie #209: 3G/4G USB Card Support

cyberhomie, 2015-06-02:
Is there any way to add 3G/4G USB Card support to MotionEye?

ccrisan, 2015-06-02:
I believe you mean motionPie (not motionEye). There are two reasons why 3G/4G modems were not integrated with motionPie:
1. most of the USB modems have that horrible CD support and require some USB modeswitch voodoo to work; from my experience, when the voodoo succeeds, the modem works 90% of the boots
2. mobile operators most often assign private IP addresses to modems and therefore your camera won't be accessible from the Internet

I have been working on some ppp integration and will hopefully include ppp/modem support some day, but I haven't had the time and modems to test what I have so far.

Squal78, 2015-06-13:
Hi all,
For the second point, it could be resolved by creating a VPN with a private network. Could it be possible to include this option in motionpie ?

ccrisan, 2015-06-14:
I will consider including OpenVPN, but I'm afraid it will generate too many questions.