ccrisan/motionpie #211: cant change Frame Change Threshold

cdnr1, 2015-06-03:
when i change Frame Change Threshold it won't detect motions any more and stops making picture stils 
when i change it from the defualt setting it stops working even if i set it back at the defualt setting
i have a raspberry pi 2 and a pi cam

ccrisan, 2015-06-04:
Alright, I'll look into it and see if I can reproduce it.

ccrisan, 2015-06-07:
I cannot seem to reproduce your problem. Does it happen all the time? Can you reproduce it?

oopeekoo, 2015-06-10:
My observations about frame change treshold setting:
- You have to reboot after every change, otherwise it's not applied.
- Setting less than 10%: too sensitive, more than 10%: almost nothing is detected.
Mask file seems to be the only way to keep false detections minimun AND detect important stuff. Too bad it's not configurable directly from Gui.

ccrisan, 2015-06-10:
@oopeekoo the mask UI is on the way. Regarding the reboot requirement, I find it hard to believe this, given the way the entire mechanism is designed. How did you test the threshold? Did you enable the Show Frame Changes setting? Did you watch the frame color (switching between red and black)? Did you analyze the movies or pictures created by motion detection?

oopeekoo, 2015-06-10:
Scenario:
False detections keep coming (shadows, tree leafs...) -> change treshold from about 8% to 12% -> hit apply -> false detections keep coming -> reboot from Gui -> no more false detections. Same other way around. Sorry, not done proper testing just day to day use. 
And it's very hard to find sweet spot (minimize false / maximize real) with this setting alone.

ccrisan, 2015-06-10:
The reboot (or eventually a proper shut down) could only affect the camera settings that are probably kept in memory. Unfortunately there's not much I can do about that. What I can do is to implement the mask feature to help reducing the false detections.

closed 2015-06-10 by ccrisan