ccrisan/motionpie #212: Securing HTTP traffic?

github-0, 2015-06-07:
Hi,

Any plans for adding HTTPS support for the web server? 

And since we're talking security - in general, would you recommend opening this to the internet? I'm fairly confident in keeping my own Pi web servers online since I can easily update the software. With projects like MotionPie (or any other 3rd party commercial/non-commercial product) I'm putting my faith on the developer to keep up with security updates (I don't have the competence/time to dig into the code). Not at all saying you wouldn't do a good job, but hopefully you understand what I mean :)

PS. THANK YOU for making this project happen, I'm in the process of adding camera surveillance to my house and this has been an awesome solution. Donation underway :)

ccrisan, 2015-06-08:
Hi,

I also usually put my fate in the hands of many FOSS developers for most of the software I use; I totally understand your worries. To be honest, I never had the chance (nor time) to test the security aspects of motionPie. I normally assume that the user only forwards port 80 when they need access from the Internet.

When it comes to securing the traffic, here are the main reasons why I never considered HTTPS for motionPie:

1. HTTPS requires certificates that cost money. Being your own certificate authority is a possibility, but then users would complain about browser warnings.
2. Obtaining the credentials using sniffing or MITM is impossible - the password is never sent over the network. There's a signature algorithm involved.
3. Replay attacks are also impossible given that each video frame has a timestamp in its URL and that the signature includes this timestamp.
4. Sniffing the traffic normally requires you to visit motionPie from a public location, next to a person that wishes to break into your house and has the necessary IT knowledge to do so. This is quite improbable.
5. In larger deployments, multiple motionPie-based devices can run behind a *reverse proxy* (using nginx or apache, for example) that takes care of HTTPS.
6. Passing the traffic through nginx + HTTPS locally (on the device) would significantly decrease the performance.

In conclusion, HTTPS is not a planned feature but can be achieved using an nginx-based reverse proxy, on which I plan to write some documentation to make the deployment easier.

Thank you very much for your donation, by the way!

closed 2015-06-08 by ccrisan

github-0, 2015-06-08:
Thanks for the explanation, I find it very reasonable. 

The information about signature algorithm is interesting, can you briefly explain how this works? Is this your own implementation or some standard component?

Reverse proxy sounds like a good solution.

ccrisan, 2015-06-09:
The signature algorithm is quite simple: it takes the request method, the URI, the body and the password and creates a big string. The string is hashed using SHA1 and that's the signature.

[This](https://bitbucket.org/ccrisan/motioneye/src/4944a504f49c2db83f5b1948e8ed3b45d4645598/src/utils.py?at=master#cl-385) is the server implementation (python) and [this](https://bitbucket.org/ccrisan/motioneye/src/4944a504f49c2db83f5b1948e8ed3b45d4645598/static/js/main.js?at=master#cl-142) is the client implementation (javascript).